stm32duino/Arduino_Core_STM32 #1126: After remapping SPI pins to alternate on STM32F103C8 and using I2C (call Wire.begin()) SPI don't work

vitmailru, 2020-07-20:
Example for reprodusing:

`#include <SPI.h>
#include <SD.h>
#include <Wire.h>

// set up variables using the SD utility library functions:
Sd2Card card;
SdVolume volume;
SdFile root;

// change this to match your SD shield or module;
// Arduino Ethernet shield: pin 4
// Adafruit SD shields and modules: pin 10
// Sparkfun SD shield: pin 8
// MKRZero SD: SDCARD_SS_PIN
const int chipSelect = PA15;

void setup() {
SPI.setMOSI(PB5);
SPI.setMISO(PB4);
SPI.setSCLK(PB3);

Wire.begin();

// Open serial communications and wait for port to open:
Serial.begin(9600);
while (!Serial) {
; // wait for serial port to connect. Needed for native USB port only
}

Serial.print("\nInitializing SD card...");

// we'll use the initialization code from the utility libraries
// since we're just testing if the card is working!
if (!card.init(SPI_HALF_SPEED, chipSelect)) {
Serial.println("initialization failed. Things to check:");
Serial.println("* is a card inserted?");
Serial.println("* is your wiring correct?");
Serial.println("* did you change the chipSelect pin to match your shield or module?");
while (1);
} else {
Serial.println("Wiring is correct and a card is present.");
}
}

void loop(void) {
}`

If comment Wire.begin() all works fine, but with Wire.begin() SPI devices do not work.

VVESTM, 2020-07-21:
For now, I understand this is more a forum question than an issue. You can visit https://www.stm32duino.com/index.php and ask help from community.

vitmailru, 2020-07-21:
I have no  questions to forum. I found bug in library and created a report

vitmailru, 2020-07-21:
if  bug  is not an issue for you then you can close it and forget it

вт, 21 июл. 2020 г. в 10:30, Виталий Гончар <vitmailru4@gmail.com>:

fpistm, 2020-08-13:
Hi @vitmailru 
sorry for the delay.
I've checked your issue and can confirm it.
Anyway, it is not linked to the core but to a device limitation, as described in the erratasheet:
https://www.st.com/resource/en/errata_sheet/cd00190234-stm32f101x8b-stm32f102x8b-and-stm32f103x8b-mediumdensity-device-limitations-stmicroelectronics.pdf

[image]

I've checked by simply disabling the I2C1 clock:
```C++
  SPI.setMOSI(PB5);
  SPI.setMISO(PB4);
  SPI.setSCLK(PB3);
  Wire.begin();
  __HAL_RCC_I2C1_CLK_DISABLE();
```

closed 2020-08-13 by fpistm