stm32duino/Arduino_Core_STM32 #1132: Wider EEPROM variable size

HamzaHajeir, 2020-07-25:
Hi

I've read EEPROM library README, It has been developed for the limited AVR architecture.

Which it's not the case for many of STM32 MCUs.

STM32F103 Flash content is 32bit size, and HAL emulation can support 16bit address / 16bit value. 

But EEPROM class is constrained to 8bit address / 8bit value, Where it leaves 16 bits of memory untouched.

I know there's many less powerful workarounds achieving the goal, But I strive for better.
[Helpful Application Note](https://www.st.com/resource/en/application_note/cd00165693-eeprom-emulation-in-stm32f10x-microcontrollers-stmicroelectronics.pdf)

fpistm, 2020-09-10:
Hi @HamzaHajeir 

I thought I've commented on this. 😕 
The main goal of the core is to be fully Arduino compatible (API) that's why it is 8 bits. EEPROM emulation handles only bytes:
https://www.arduino.cc/en/Reference/EEPROM

So the built in EEPROM library should not be changed. Maybe it can be extended or simpliest way would be to implement an new one named for example: EEPROM16.

Feel free to contribnute and submit a PR. 😉

ajlajlajl, 2020-11-17:
Hi
I was looking around to see if there has been any effort to expand current EEPROM lib, to be more, dynamic. anyway. I saw this issue.
I think what @HamzaHajeir is saying is not related to your implementation of EEPROM. 
If I'm not wrong, what HamzaHajeir is talking about is what described as EEPROM emulation in AN2594 (and AN4894) that he mentioned. but that method was not used in this implementation. so "Wider variable size" or "16 bits of memory untouched" has nothing to do to this EEPROM lib. (and really, no matter ho i look at that, it was not a good way, it has fixed sized variables and half of memory is wasted).

I think I should open another issue related to expanding current implementation, as I had to implement my own as its too limited on this one. I'll study a little more about this for now.

fpistm, 2020-11-17:
Hi @ajlajlajl 
yes you're right. It is more related to AN as it is possible to store all desired data as shown in default example.

> I think I should open another issue related to expanding current implementation, as I had to implement my own as its too limited on this one. I'll study a little more about this for now.

As mentioned:

> The main goal of the core is to be fully Arduino compatible (API)

You can expand the implementation while it is backward compatible.

closed 2020-11-17 by fpistm