stm32duino/Arduino_Core_STM32 #1134: Add support for WeAct HID bootloader v1.x

pawelsky, 2020-08-01:
[WeAct STM32F401CE and STM32F411CE](https://github.com/WeActTC/MiniF4-STM32F4x1) boards come with a modified HID bootloader (with HID device ID 0483:572A) therefore it is not possible to upload the code with standard HID 2.2 Bootloader (expecting HID device ID 1209:BEBA).

Please add support for [WeAct HID Bootloader 1.x](url). It can be added as yet another bootloader option for Generic F4 variants. Alternatively separate WeAct F4x1CE variants can be created.

samveen, 2020-08-06:
@pawelsky The WeAct bootloader also modifies the mechanism used to trigger the bootloader process (I was trying to get the it to work for my board (see #1118). Not only that, the WeAct folks also took the stm32HIDBootloader, and modified it a bit too (in a similar manner, changing the bootloader check). I had to do a bit of flailing but I managed to get a working version that works with arduino IDE [here](https://github.com/samveen/STM32_HID_Bootloader/tree/merged)

pawelsky, 2020-08-06:
@samveen Correct, they did make some more modification, and yes you can change the bootloader on the board, but the goal of this feature request is to have the WeAct bootloader supported in the IDE (even if it means manually putting the board into the bootloader mode), so people who bought WeAct boards with their bootloader do not have to replace it.

fpistm, 2020-08-13:
Hi @pawelsky 

> Please add support for [WeAct HID Bootloader 1.x](https://github.com/WeActTC/MiniF4-STM32F4x1/blob/master/Soft/WeAct_HID_FW_Bootloader/README.md). It can be added as yet another bootloader option for Generic F4 variants. Alternatively separate WeAct F4x1CE variants can be created.

There are more and more fork of the HID bootloader and I could not add all of them to not increase the menu and this will add more confusion.
Moreover user can add his own upload menu by adding a platform.local.txt and a boards.local.txt.

closed 2020-08-13 by fpistm

pawelsky, 2020-08-13:
> Moreover user can add his own upload menu by adding a platform.local.txt and a boards.local.txt.

Yes, I know you can, but the whole point of this ticket was to have it integrated with official releases so people don't have to tinker themselves.

Anyway, for those who want to use it attached is a package that adds the WeActHIDBootloader support. All you need to do is to unpack its contents to your `...\packages\STM32\hardware\stm32\x.y.z` directory and restart the Arduino IDE.
[WeActHIDBootloader_20200813.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/5071572/WeActHIDBootloader_20200813.zip)

You should then have the WeAct HIB Bootloader upload option available

[image]

Note that if you already have your boards/platforms local files there, you should merge their contents with the ones from zip file.

Only tested on Windows but it should also work on Linux