stm32duino/Arduino_Core_STM32 #1144: Add USER_BTN mapping to PA0 for BlackPill F401CC/F401CE boards

pawelsky, 2020-08-15:
Since v1.2 BlackPill F401CC/F401CE boards have an user button (named "KEY") connected to PA0 added to the board. You can find more details about this board [here](https://github.com/WeActTC/MiniF4-STM32F4x1)

Please add the USER_BTN define mapping to PA0 for BlackPill F401CC and the [proposed F401CE](https://github.com/stm32duino/Arduino_Core_STM32/issues/1143) boards. That would be consistent with the already existing [BlackPill F411CE USER_BTN mapping](https://github.com/stm32duino/Arduino_Core_STM32/blob/59dda5c138badcb0a17a24f7a30fa81dececa0d7/variants/Generic_F411Cx/variant.h#L83)

I believe the following change in the [Generic_F401Cx/variant.h](https://github.com/stm32duino/Arduino_Core_STM32/blob/59dda5c138badcb0a17a24f7a30fa81dececa0d7/variants/Generic_F401Cx/variant.h#L82) file should do the job:

> // On-board user button
> #if defined(ARDUINO_BLACKPILL_F401CC) || defined(ARDUINO_BLACKPILL_F401CE)
> #define USER_BTN                PA0
> #elif !defined(USER_BTN) && !defined(ARDUINO_GENERIC_F401CC)
> #define USER_BTN                PC13
> #endif

ABOSTM, 2021-04-02:
Hi @pawelsky,
The BlackPill F401CC doesn't have KEY button.
The BlackPill F401CE have the button but this board is not yet supported.
So this issue correspond to the request to support BlackPill F401CE and thus is duplicate of #1143
BlackPill F401CE has already been add to the request list #722.
This is a community project, so any contribution to add this board is welcome (even if I would suggest to way for PR1091 to be merged first).
Thus I close this issue.

closed 2021-04-02 by ABOSTM

pawelsky, 2021-04-02:
There are versions of F401CC that do have the key (photo of v3.0 attached), so I'd request this issue to be reopened.

[image]

ABOSTM, 2021-04-02:
Ok I was not aware of this change in latest board revision.
Even if it could be considered as a different board, we decided to consider only 1 board in which USER_BTN is defined,
even if user button is only available on some board revisions.
It is implemented there:
https://github.com/fpistm/Arduino_Core_STM32/pull/12/commits/72dd95fcc6c91b9ac94c0b97234161ff85554a8b
And it will be part of #1091 
(thus no need to reopen this issue)

pawelsky, 2021-04-02:
That makes sense. Thanks!