stm32duino/Arduino_Core_STM32 #1145: Support for MCUDev DevEBox STM32H7XX_M board (STM32H750VBT6)

observator-x, 2020-08-16:
- Name: MCUDev DevEBox STM32H7XX_M
 - Link to all relevant information:
  https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M

MCUDev STM32H7XX_M mini dev boards.

There are two versions with different STM32H7 chips:

STM32H750VBT6 - 128K flash
STM32H743VIT6 - 2048K flash

Markings: DevEBox STM32H7XX_M Ver:V2.0 SN:1907

fpistm, 2020-08-17:
Added to the [New Variant] request list #722.

closed 2020-08-17 by fpistm

Hedda, 2021-02-12:
Please note that there looks to be two variants of MCUDev DevEBox STM32H7XX_M boards with different chips / flash memory.

MCUDev DevEBox STM32H7XX_M with STM32H750VBT6 (128KB flash memory & 1024KB SRAM)

MCUDev DevEBox STM32H7XX_M with STM32H743VIT6 (2048KB flash memory & 1024KB SRAM)

https://www.cnx-software.com/2021/01/23/devebox-stm32h7-development-boards-are-made-for-factory-automation/

http://myosuploads3.banggood.com/products/20200410/20200410041012H750VI-.rar
http://myosuploads3.banggood.com/products/20200410/20200410034233H743VI-.rar

https://github.com/elzoughby/STM32H7xx-Startup

ag88, 2021-11-21:
note that the variant made for WeAct MiniSTM32H743VITX would likely work for MCUDev DevEBox STM32H743VIT6
https://github.com/stm32duino/Arduino_Core_STM32/pull/1552
As it uses 25 Mhz HSE crystal same on both boards.
However, do note that the symbols LED_BUILTIN and USER_BTN are different between the 2 boards.
To use the WeAct variannt for MCUDev DevEBox STM32H743VIT6, avoid using the 2 symbols or re-define them in your arduino sketch. you could also simply clone the WeActH743VITX variant file and make mods locally e.g. make a new variant if you prefer.

PR #1573 (2021-11-30): variants: add DevEBox H743VITX and H750VBT support
https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M

Fixes #1145 and fixes #1266
commit f5c6e7ac (2021-11-30): variants: add DevEBox H743VITX and H750VBT support

https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M

Fixes #1145 and fixes #1266