stm32duino/Arduino_Core_STM32 #1146: [STM32F070]

alberiolima, 2020-08-19:
I tested it on the stm32F070 using the "STM32F030 DEMO BOARD", replacing the STM32F030F4P6 with an STM32F070F6P6. Apparently it works normally!

# DEMO_F070F6 board
GenF0.menu.pnum.DEMO_F070F6=STM32F070F6 Demo board
GenF0.menu.pnum.DEMO_F070F6.upload.maximum_data_size=6144
GenF0.menu.pnum.DEMO_F070F6.upload.maximum_size=32768
GenF0.menu.pnum.DEMO_F070F6.build.board=DEMO_F070F6
GenF0.menu.pnum.DEMO_F070F6.build.product_line=STM32F070x6
GenF0.menu.pnum.DEMO_F070F6.build.variant=DEMO_F030F4

alberiolima, 2020-08-19:
now i will check pins and serial ports to modify "variants"

fpistm, 2020-08-19:
Hi, thanks for the info.
I'm currently working to ease generic series/boards addition.

fpistm, 2020-08-19:
Added to the new variant request list #722

closed 2020-08-19 by fpistm