stm32duino/Arduino_Core_STM32 #1147: [New Variant]

AliBigdeli, 2020-08-19:
**Board description (please complete the following information):**
 - Name: Nucleo64 STM32G070
 - Link to all relevant information:
   * Documentations : https://www.st.com/resource/en/user_manual/dm00452640-stm32-nucleo64-boards-with-stm32g07xrb-mcus-stmicroelectronics.pdf
   * Schematics :https://www.manualslib.com/manual/1529028/St-Stm32-Nucleo-64.html?page=11
   * Where this hardware can be purchased :https://www.st.com/en/evaluation-tools/nucleo-g070rb.html#:~:text=The%20STM32%20Nucleo%2D64%20boards,provided%20by%20the%20STM32%20microcontroller.

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

All new variant request will be moved to [New Variant] request list #722 and the related issue will be closed.

fpistm, 2020-08-19:
Hi,

don't know which site you talk about. Only the Nucleo G071RB exists not the G070 one.

fpistm, 2020-08-19:
Added to the new variant request list #722

closed 2020-08-19 by fpistm

fpistm, 2022-06-21:
Implemented by #1745

PR #1745 (2022-06-21): variant: G0: add Nucleo-G070RB
[Nucleo_g070rb
](https://www.st.com/en/evaluation-tools/nucleo-g070rb.html)
### **Tested:**
 

- GPIO
- U(S)ART
- ADC
- DAC
- I2C
- SPI
- Timer
- RTC
- Low Power
- EEPROM

Fixes #1147 and fixes #1651