stm32duino/Arduino_Core_STM32 #1149: I2C STM32F4 - Question

LDaveX, 2020-08-19:
Hi,

I have been working with a [WeActTC STM32F401CC V3.0](https://github.com/WeActTC/MiniF4-STM32F4x1) for a while and I am not able to make the I2C interface to work. 

I have tried the following code as an example with different boards (Arduino Uno, W600, BluePill STM32F103C8T6) and the code works without any problem. I can see the I2C requests with a Logic Analyser. 

```
#include <Arduino.h>
#include <Wire.h>

void setup() 
{
    Wire.begin();
}
 
void loop() 
{
    volatile uint32_t data = 0;

    Wire.requestFrom(8,1); 
    delay(30);

    data = (uint32_t) Wire.read();   
    delay(100);
}
```
Regarding the WeActTC STM32F401CC board, the Logic Analyser is connected to SDA1 and SCK1 which are PB7 and PB6, according to [WeActt repository](https://github.com/WeActTC/MiniF4-STM32F4x1/blob/master/images/STM32F4x1_PinoutDiagram_RichardBalint.png). I have tried Arduino IDE as well as PlatformIO.

Probably I am just not doing something properly, but so far I have not been able to find out what it is. Any piece of advice would be appreciated.

Thank you so much.

Regards,
David.  

**Desktop:**
 - OS: Windows
 - PlatformIO: 4.3.4
 - Arduino IDE version: 1.8.12
 - STM32 core version: 8.0.0

**Board:**
 - Name: WeActTC STM32F401CC
 - Hardware Revision: V3

fpistm, 2020-08-19:
Hi @LDaveX 

For question please use the forum: https://www.stm32duino.com/

As a first answer, did you use Pull up resistors on each I2C lines ?

fpistm, 2020-08-19:
Just a quick note:

> STM32 core version: 8.0.0

This is not the core version, if you check the release, the latest is 1.9.0. I guess this is the PIO one but this is not supported here.

closed 2020-08-19 by fpistm