stm32duino/Arduino_Core_STM32 #1151: Not seeing characters out of serial port, (STM32F030F4P6 v2.0)

RandoSY, 2020-08-20:
Working with simple serial port output test this morning.
 
I was wondering if the library was writing to the correct pins from here [https://stm32-base.org/boards/STM32F030F4P6-STM32F030F4-V2.00](url) , in this case PA9. I'm not seeing any serial characters transmitted to my USB2TLL adapter. 

Serial programming works perfectly on another test on this chip using the same setup, and the LED tell tale is blinking happily, when I run the example 

-- | -- | -- | --
1 | 3V3 | - | +3.3V rail
2 | RX | - | PA10
3 | TX | - | PA9
4 | GND | - | Ground plane

CODE FOLLOWS

`void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  while(!Serial);
}

void printLoop() {

   for (uint16_t i = 0; i < 4096; i++)
  {
      Serial.println("TEST_RSY");
      //Explicitly flash alert that string has been printed
      digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
      delay(500);                       // wait for a second
      digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage L
      delay(500);
  }
}

//main loop 
void loop() {
  // put your main code here, to run repeatedly:
  // add blinky output in printLoop()
  printLoop();
  delay(100);  
}
`

Suggestions?

fpistm, 2020-08-20:
Hi @RandoSY 

The DEMO F4 referenced in the core define Serial to be on

RandoSY, 2020-08-20:
Thank you for the consideration of your response. :)

fpistm, 2020-08-20:
Welcome,
FYI https://github.com/stm32duino/wiki/wiki/API#hardwareserial