stm32duino/Arduino_Core_STM32 #1158: gsm with serial1 and stm32f103c8t6

laxman432, 2020-08-31:
hi i am using arduino ide for programming recently we updated ide and stm32arduino library  we written just message sending AT commands it is executing but not getting messages. when i uploaded old version(before updation ) binary file  i am getting messages but when i uploaded new compiled binary not able to recive sms we need to know it is very urgent if anyone know that problem please help me we tried so many times 

here is the code 

 #define gsm serial1

void setup()
{
  gsm.begin(9600);   // Setting the baud rate of GSM Module  
  Serial.begin(9600);    // Setting the baud rate of Serial Monitor (Arduino)
  delay(100);
}

void loop()
{
  if (Serial.available()>0)
   switch(Serial.read())
  {
    case 's':
      SendMessage();
      break;
    case 'r':
      RecieveMessage();
      break;
  }

 if (gsm.available()>0)
   Serial.write(gsm.read());
}

 void SendMessage()
{
gsm.println("AT+CMGF=1");    //Sets the GSM Module in Text Mode
  delay(1000);  // Delay of 1000 milli seconds or 1 second
  gsm.println("AT+CMGS=\"+918143710109\"\r"); // Replace x with mobile number
  delay(1000);
gsm.println("I am SMS from GSM Module");// The SMS text you want to send
  delay(100);
gsm.println((char)26);// ASCII code of CTRL+Z
  delay(1000);
}

 void RecieveMessage()
{
  gsm.println("AT+CNMI=2,2,0,0,0"); // AT Command to receive a live SMS
  delay(1000);
 }

fpistm, 2020-08-31:
Hi @laxman432 
Could you be more precise about version?
And also about the variant you used? Board?
All relevant information.

laxman432, 2020-08-31:
[image: image.png]

*                                  Thanks & Regards,*
                                  * M.P.LAXMAN M.tech (Ph.D.)*
*                            CONTACT: **8143710109 *

laxman432, 2020-08-31:
hi i checked with latest versions only i am not able to recieve sms from my hardware here we used arduino 1.8.13 stm32 library 1.9.0 when i check with previas versions 2017 it was working now that 2017 lbrary not able to put it we need its very urgent help me in this

fpistm, 2020-08-31:
Sorry @laxman432 
but this is really not clear.
what is the previous version 2017 ?
Moreover what is the board you used ? gsm shield?...

laxman432, 2020-08-31:
we are using sim800c gsm ic  with stm32f103c8t6 we connected pa2  and pa3 to gsm rx and tx pins 
when i upload my previas binary file it is working perfectly at that time we used arduino 1.6.12,stm32library 2017.9.21 
now we updated our software it is compiled and it uploaded in to board but i am not getting any messages no communication created with updated library please help me it is very urgent

laxman432, 2020-08-31:
in previas we used below process it is working properly now we are not able
to get those libraries

*                                  Thanks & Regards,*
                                  * M.P.LAXMAN M.tech (Ph.D.)*
*                            CONTACT: **8143710109 *

fpistm, 2020-08-31:
Well, which process? which libraries you are not able to get?

Note that here we spoke about the core, this is not a library. Latest version is 1.9.0. There is no 2017.9.21.
Only [2017.9.22](https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/2017.9.22) with old versioning which became 1.0.1.

fpistm, 2020-08-31:
Are you sure you used this core before ?
Maybe you used this one: https://github.com/rogerclarkmelbourne/Arduino_STM32
Using this boards package file: http://dan.drown.org/stm32duino/package_STM32duino_index.json

laxman432, 2020-08-31:
http://dan.drown.org/stm32duino/package_STM32duino_index.json

by using above link we got that 2017.9.21

now we are not able to see that core 

we generated binary file using that core now also it is working but if i  want do any modifications  not possible with updated

fpistm, 2020-08-31:
So, this is not the same core. If you have issue with https://github.com/rogerclarkmelbourne/Arduino_STM32 using the one installed thanks http://dan.drown.org/stm32duino/package_STM32duino_index.json then you have to request support there.

fpistm, 2020-08-31:
Note that this core (https://github.com/stm32duino/Arduino_Core_STM32) release 1.9.0, if you used the Generic F103C8 board then by default the Serial instance is mapped on Serial1 and used those pins by default:
```
#define PIN_SERIAL_RX           PA10
#define PIN_SERIAL_TX           PA9
```

fpistm, 2020-08-31:
If I well understand your use case:
Serial is the Serial monitor connected to the host and Serial1 is used to communicate with your GSM module connected to PA2/PA3.

In that case with this core:

```
HardwareSerial Serial2(PA3, PA2);

#define gsm serial2

void setup()
{
  gsm.begin(9600);   // Setting the baud rate of GSM Module
  Serial.begin(9600);    // Setting the baud rate of Serial Monitor (Arduino)
  delay(100);
}

void loop()
{
  if (Serial.available() > 0)
    switch (Serial.read())
    {
      case 's':
        SendMessage();
        break;
      case 'r':
        RecieveMessage();
        break;
    }

  if (gsm.available() > 0)
    Serial.write(gsm.read());
}

void SendMessage()
{
  gsm.println("AT+CMGF=1");    //Sets the GSM Module in Text Mode
  delay(1000);  // Delay of 1000 milli seconds or 1 second
  gsm.println("AT+CMGS=\"+918143710109\"\r"); // Replace x with mobile number
  delay(1000);
  gsm.println("I am SMS from GSM Module");// The SMS text you want to send
  delay(100);
  gsm.println((char)26);// ASCII code of CTRL+Z
  delay(1000);
}

void RecieveMessage()
{
  gsm.println("AT+CNMI=2,2,0,0,0"); // AT Command to receive a live SMS
  delay(1000);
}
```

With by default Serial is mapped to Serial1 which is a real USART using PA9/P10. If you select USB CDC in the menu then Serial will be the SerialUSB.

laxman432, 2020-08-31:
hi we are getting error can u give hardwareserial library link

fpistm, 2020-08-31:
which kind of error, please share a log.
You well used this core with release 1.9.0? With which options ?

laxman432, 2020-08-31:
it is working thankyou very much for your support actually we miss leaded to other core now on words we will follow this core only

closed 2020-08-31 by fpistm

laxman432, 2020-08-31:
Hi I uploaded my production code using hardware serial it is not working properly it means I stores one mobile number to get sms when I send sms after few minits that number not available again I need to make registration 

Is there any stm32core changes from old to new version

fpistm, 2020-08-31:
@laxman432 
it seems you don't understand which core you used.
The one you used before (2017 is very old) is not the same than this one.

laxman432, 2020-08-31:
We used 2017.9.21 we have document also  with that I generated binary file it is working perfectly with serial1 even we didn't use hardware serial
 With new versions it is not working properly guide me how solve this issue

fpistm, 2020-08-31:
@laxman432 
The new version as you told of the 2017.9.21 is not here. This is not the same core.
2017.9.21 is this core: https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/816

laxman432, 2020-08-31:
Why the communication is not working properly it is miss behaving

fpistm, 2020-08-31:
Well, honestly, I don't know as it works properly since a while.
I guess you have an issue on your side, mainly around the usage of the core and also about the STM32 core you used. It is not clear which one you used...

Nikitanagar, 2024-12-05:
@laxman432, I hope you had solved your issue. May I know the reason because I am also facing the same issue?