stm32duino/Arduino_Core_STM32 #1160: Support for UART RX/TX level inversion on F3/F7 platforms

pawelsky, 2020-08-31:
Please add API in HardwareSerial class to enable/disable RX/TX level inversion for platforms that support it (e.g. F3 or F7)

This can be done e.g. by adding a inverse_logic parameter to the constructor (similarily to SoftwareSerial), and based on that configuring the AdvancedInit member of the UART_AdvFeatureInitTypeDef struct in uart_init function in the uart.c wrapper to something like:
```
    huart->AdvancedInit.AdvFeatureInit = UART_ADVFEATURE_TXINVERT_INIT | UART_ADVFEATURE_RXINVERT_INIT;
    huart->AdvancedInit.TxPinLevelInvert = UART_ADVFEATURE_TXINV_ENABLE;
    huart->AdvancedInit.RxPinLevelInvert = UART_ADVFEATURE_RXINV_ENABLE;
```

babipsylon, 2020-12-03:
How to (un)invert on current firmware bf 4.16?

fpistm, 2021-06-11:
Added to [U(S)ART] request list #1418

commit 3a607767 (2025-03-31): feat(uart): support UART Tx, Rx and data invert function

This enables UART Tx, Rx and data invert function on STM32 families that support it.

In order to enable Tx, Rx and/or data invert, call respectively:

```c++
Serial1.setTxInvert();
Serial1.setRxInvert();
Serial1.setDataInvert();
```

Fixes: #1160 #2669
See also: #1418
closed 2021-06-11 by fpistm