stm32duino/Arduino_Core_STM32 #1162: HardwareSerial Error Callback

0x6d61726b, 2020-09-01:
For implementing a custom protocol handler based on UART, it is necessary to get knowledge about the LIN Break Detection,  Framing Error and Parity Error signals. What is the proposed way for such an implementation?

Would it be ok to propose an extenions of the HarewareSerial class to provide a error callback mechanism that gets called by the [HAL_UART_ErrorCallback()](https://github.com/stm32duino/Arduino_Core_STM32/blob/6fff465282dacdfc63a13bc845caf4a8ebd2b9c4/libraries/SrcWrapper/src/stm32/uart.c#L829)? If so, I would be happy to make such a proposal for further discussion.

fpistm, 2020-09-01:
Hi @0x6d61726b 
HardwareSerial follows the Arduino API that's why there is no error CB.
Anyway feel free to contribute and propose a PR, any contribution are welcome.
But pay attention to not break the standard API 😉

0x6d61726b, 2020-09-01:
Hi @fpistm, thanks for the positive feedback. :blush:
I was unsure if an extension is possible.
Not altering the standard API functionality is taken for granted.

I will come back in the next days with the proposal.

fpistm, 2021-06-11:
Added to [U(S)ART] request list #1418

closed 2021-06-11 by fpistm