stm32duino/Arduino_Core_STM32 #1172: [New Variant] Generic STM32H743VIT6

vladkozlov69, 2020-09-14:
**Board description:**
 - Name: Generic STM32H743VIT6
 - Link to all relevant information:
   * documentation: https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M
   * Schematics: https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M/blob/master/docs/STM32H7XX_M_schematics.pdf
   * This device is selled on Aliexpress, board marked as DevEBox STM32H7XX_M Ver:V2.0 SN:1907

PS: I have tried to use NUCLEO_H743ZI variant with clock initialization generated in CubeMX, got only I2C and GPIO working, SDMMC and USART hangs for the unknown reason.

fpistm, 2020-09-14:
Duplicate of #1145.
Already added in the #722

closed 2020-09-14 by fpistm

ag88, 2021-11-21:
note that the variant made for WeAct MiniSTM32H743VITX would likely work for MCUDev DevEBox STM32H743VIT6
https://github.com/stm32duino/Arduino_Core_STM32/pull/1552
As it uses 25 Mhz HSE crystal same on both boards.
However, do note that the symbols LED_BUILTIN and USER_BTN are different between the 2 boards.
To use the WeAct variannt for MCUDev DevEBox STM32H743VIT6, avoid using the 2 symbols or re-define them in your arduino sketch. you could also simply clone the WeActH743VITX variant file and make mods locally e.g. make a new variant if you prefer.