stm32duino/Arduino_Core_STM32 #1178: Remove Arduino Library Overhead

shuzonudas, 2020-09-20:
Hi,
I would like to know if it is possible to compile stm32f030 chip without the Arduino library overhead. When compiling for stm32F030, an empty sketch is around 9kb. If there is an option to exclude all the Arduino overhead, it would be cool.

fpistm, 2020-09-20:
You can try to disable unused HAL module. https://github.com/stm32duino/wiki/wiki/HAL-configuration#hal-modules-configuration

Other option is to use the LL for RCC config by overriding the system core clock config.

shuzonudas, 2020-09-20:
Is it possible to configure "HAL modules" for individual projects?
Like in project one I want to disable to RTC and I2C
but in project two I would like to disable ADC only.

fpistm, 2020-09-20:
Yes  as hal_conf_extra.h or build_opt.h are per project. 

Please for this kind of question or support use the forum.

closed 2020-10-06 by fpistm