stm32duino/Arduino_Core_STM32 #1179: I2C stops after heavy load

saloid, 2020-09-20:
**Describe the bug**
I2C suddenly stops working after some time of heavy load - continuous reading data from BMP280 and MPU9250. This can happen in 1-30 seconds after start. I'm using PB10 and PB11 pins, at 400kHz speed.
Data transfer always stops at the same packet (reading answer (is data ready) from BMP280).
[image]

**To Reproduce**
Here is my [platformio project](https://github.com/giver0/Rocket-logger/tree/dev/Soft/Embedded/Main-program)

**Screenshots**
No I2C activity after this issue happens:
[image]

Last packet. It looks like others "good" packets:
[image]
Good packet:
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - STM32 core version: in platformio it's 8.1.0
 - Upload method: STLink-V2

**Board (please complete the following information):**
 - Name: custom PCB with STM32F103RCT6

saloid, 2020-09-20:
Ok, looks like it's [this](https://github.com/MartinL1/BMP280_DEV) BMP280 library issue - if i will leave it alone, it will also fail. Another library ([from adafruit](https://github.com/adafruit/Adafruit_BMP280_Library)) works fine.

fpistm, 2020-09-21:
Hi @saloid 
if this is linked to the library then we could not help on this.
So I close this.
Thanks for your feedback about library issue.

saloid, 2020-10-18:
It's not library issue - it happens on different libraries, just need to wait some time. All of them using the same arduino functions:
```
Wire.beginTransmission();         
Wire.write();                  
Wire.endTransmission();             
Wire.requestFrom();	 
data = Wire.read();
```
When fail occurs, I have noticed that STOP bit in I2C2->CR1 is always set and newer cleared. Due to this, next i2c transaction newer happens, it just exits after timeout with error code at [this place](https://github.com/stm32duino/Arduino_Core_STM32/blob/6fff465282dacdfc63a13bc845caf4a8ebd2b9c4/system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_i2c.c#L3581).

Reopen this issue, please. (or should I create new one?)

fpistm, 2020-10-20:
Hi @saloid 
Could you try with the master branch, they are I2C fixes linked to this.

saloid, 2020-10-31:
It still happens, but much more rarely and with shorter timeouts at faulty transfer (before it was about 1 second and now it's almost instantly returns). At this time SCL line sometimes holded down for 10 ms. And as before, last successful transfer looks completely valid - same to all previous transfers. I have tried 400 and 100 kHz - same result.

SCL holded down:
[image]

SCL with short bursts (<250ns)
[image]

Last transfer:
[image]

saloid, 2020-10-31:
`Wire.endTransmission()` and `Wire.requestFrom()` always returning 0 (success)

saloid, 2020-10-31:
@fpistm, could you please reopen this issue? I am gonna try it on bluepill on different I2C channels (currently I have blue pill with chinese STM32 clone, but I can find original if it matters).

ABOSTM, 2020-11-02:
Hi @saloid,
Yes, testing a bluepill with original STM32 is necessary as it may be an hardware issue.

As I understand, problem occurs when you have both BMP280 and MPU9250,
but failure always occurs on BMP280, right? Meaning previous good I2C transaction was on BMP280.
Unfortunately I have none of those devices to test.
So can you tell me:
* Is issue reproducible when only BMP820 is connected ?
* Is issue reproducible when using MPU9250 + another I2C device ? (If you have another I2C device than BMP280 to connect in addition to MPU9250)

> When fail occurs, I have noticed that STOP bit in I2C2->CR1 is always set and newer cleared.

This is in contradiction to I2C register dump 'After entering into faulty state':
CR1 = 0x101 ==> Bit 'START' and bit 'Peripheral enable' are set.

Also SR2 = 0x2 ==> Busy flag is set.
[image]
So Busy flag maybe explained by SCL Low.
SCL may be maintained low by slave I2C. It is called 'stretching the clock'. It is used for example when master request to read some data to the slave, Master send write command with the address of the data to read (for example), and then slave stretch the clock in order to prepare the data to be send, and finally slave send data (and release CLK)
This is the case in the last logical analyser "Last transfer". there is write operation then read. And in between, the clock is streched low. This plot seems to end properly, there is a stop condition generated and both line are pulled up at the end.
It seems different on other plots, "SCL holded down" seems to occurs quite 'long' time after the end of previous transfert ends.I don't know whether it ends with STOP condition, I need some zoom in to see scl and sda transition, just before SCL is stretched low for some times, and also when there is short burst on scl.

saloid, 2020-11-02:
Problem occurs when I am trying to communicate with BMP280 alone. MPU9250 is on same I2C bus, but it even not inited and newer accessed (I can desolder it if it can cause issue).

ABOSTM, 2020-11-02:
No need to desolder MPU9250. But I am still interested by other tests described above.
In addition, can you also try to replace stm32f1xx_hal_i2c.c by this file (remove '.txt' the extension)
[stm32f1xx_hal_i2c.c.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/5476198/stm32f1xx_hal_i2c.c.txt)
and tell me if you can reproduce the issue, thanks.

ABOSTM, 2020-11-02:
Sorry I put the wrong file, use this one instead:
[stm32f1xx_hal_i2c.c.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/5476326/stm32f1xx_hal_i2c.c.txt)

saloid, 2020-11-12:
Small update: on chinese BluePill with not original STM32F103C8, I2C works pretty stable (couple of days without any issue).

saloid, 2020-11-22:
Thank you for all help! It was hardware issue - BMP280 sometimes was starting to hold SCL line in HIGH state, so any I2C activity was not possible after this. I have soldered another chip and now all works as it should.

closed 2020-11-22 by saloid

fpistm, 2020-11-23:
Thanks for the feedback.

mrunal-shade, 2020-11-30:
I am facing the same problem with mpu6050 . I dont know what is causing trouble ?

saloid, 2020-11-30:
@mrunal-shade, try latest version from master. In platformio:
'platform_packages = framework-arduinoststm32 @ https://github.com/stm32duino/Arduino_Core_STM32'

mrunal-shade, 2020-12-01:
@saloid i figured it out . I reduced the i2c clock frequency from 400kHz to 70kHz and it worked fine. After reading the datasheet of stm32 under i2c it was given .... To achieve communication at higher clock speeds you need to add some pullups around 5k to vcc and to the scl and sda . So i didnt want higher speed as my mpu6050 is sampling data at 1khz . So i went with lower speed. Thankyou so much for your help !

saloid, 2020-12-01:
I am working on 400kHz with 1.5k on my custom board and 4.7k on demo board with generic MPU9250/MPU6050. Both works ok.