stm32duino/Arduino_Core_STM32 #1180: UART1 on pin PA10, PA9 does not work on Nucleo-G071RB

harji2130, 2020-09-21:
I found issue where UART1 is not working on pin **PA10 & PA9** on **Nucleo-G071RB** but UART1 works on pin **PB7 & PB6**.
 
I verified with following code in Arduino IDE and PlatformIO

 ```c
  

HardwareSerial rfUart(PA10, PA9);  // do not work
//HardwareSerial rfUart(PB7, PB6); // work successfully
  
void setup()
{
  Serial.begin(115200);
  delay(100);

  rfUart.begin(115200);

  pinMode(LED_BUILTIN, OUTPUT);

  Serial.println("\nSG0 Begin....");
  rfUart.println("\nSG0 RF Begin....");
}

int count = 0;
void loop()
{
  delay(2000);

  Serial.printf("\nSG0 Count: %02d", count);
  rfUart.print("\nRF Count:"); rfUart.println(count++);

  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
}
```

**Note:** I tried using Stm32Cube Framework, UART1 on pin PA10 & PA9 works fine.

fpistm, 2020-09-22:
Hi @harji2130 

Thanks for reporting this issue. Error comes from the REMAP of those pins. When adding the Disco G031 we introduced the remap by default but this is not correct.

https://github.com/stm32duino/Arduino_Core_STM32/blob/95e58e64f8d997a18806abd5ac7e6fd7d9544ff5/libraries/SrcWrapper/src/stm32/pinmap.c#L75-L96

So in your case PA9 is remap to PA11. If you wire the Tx pin on PA11 you will see the output.

I will fix that soon.

harji2130, 2020-09-22:
Thank you @fpistm, for now as workaround, I commented out the defines in **"stm32g0xx_ll_system.h"** 

```c
//#define LL_SYSCFG_PIN_RMP_PA11              SYSCFG_CFGR1_PA11_RMP                           /*!< PA11 pad behaves as PA9 pin */
//#define LL_SYSCFG_PIN_RMP_PA12              SYSCFG_CFGR1_PA12_RMP                           /*!< PA12 pad behaves as PA10 pin */
```

harji2130, 2020-09-24:
Hello @fpistm , I was further exploring more uart for my project and found that UART4 on pin PA1 & PA0 also do not work under Arduino framework.
 
I have tested the UART1, UART2 & UART4 using STM32Cube Framework and everything works fine. 

```c
#include <Arduino.h> 

#define pin  LED_BUILTIN

HardwareSerial UART1(PA10, PA9);          // works fine after workaround provided by @fpistm in above comments
HardwareSerial Uart4(PA1, PA0);             // UART 4 on Pin PA1 & PA0 does not work and MCU stop working.
UART_HandleTypeDef huart4;                 // HACK using STM32 HAL Library  for UART4 on Pin PA1 & PA0

void Uart4_Init();
void UART4_Send(); 
 
void setup()
{
  Serial.begin(115200);               // on Pin rx_PA3 tx_PA2
  delay(100);

  UART1.begin(115200);             // on Pin rx_PA10 tx_PA9 
  delay(100);

  Uart4.begin(115200);               // Error working using Arduino library // on Pin rx_PA1 tx_PA0
  //Uart4_Init();                           // using STM32 HAL Library as workaround to send data but cannot receive due to lib conflict
  delay(100);

  pinMode(LED_BUILTIN, OUTPUT);
  
}

int count = 0;
void loop()
{
  delay(1000); 

  Serial.print("\nSG0  UART2(Serial): "); Serial.print(++count);
  
  UART1.print("\nSG0 UART1: "); UART1.print(count);
  
  Uart4.print("\nSG0 Uart4: "); Uart4.print(count); // **SOC stop working at this point...**
  
  //UART4_Send(); //using STM32 HAL Library (Sending is fine, receiving data is problem)
 
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
}

/* Sending data on UART 4 using STM32 HAL Library*/
void UART4_Send()
{
   char str[20];
   String thisString = "\nSG0  UART4: " + String(count) + "\n";
   thisString.toCharArray(str, thisString.length());
   HAL_UART_Transmit(&huart4, (uint8_t*)str, thisString.length()-1, 1000);
}

/* Initializing UART 4 using STM32 HAL Library*/ 
void Uart4_Init()
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  /* Peripheral clock enable */
  __HAL_RCC_USART4_CLK_ENABLE();

  __HAL_RCC_GPIOA_CLK_ENABLE();
  /**USART4 GPIO Configuration
  PA0     ------> USART4_TX
  PA1     ------> USART4_RX
  */
  GPIO_InitStruct.Pin = GPIO_PIN_0|GPIO_PIN_1;
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_LOW;
  GPIO_InitStruct.Alternate = GPIO_AF4_USART4;
  HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

     /* USER CODE END USART4_Init 1 */
  huart4.Instance = USART4;
  huart4.Init.BaudRate = 115200;
  huart4.Init.WordLength = UART_WORDLENGTH_8B;
  huart4.Init.StopBits = UART_STOPBITS_1;
  huart4.Init.Parity = UART_PARITY_NONE;
  huart4.Init.Mode = UART_MODE_TX_RX;
  huart4.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart4.Init.OverSampling = UART_OVERSAMPLING_16;
  huart4.Init.OneBitSampling = UART_ONE_BIT_SAMPLE_DISABLE;
  huart4.Init.ClockPrescaler = UART_PRESCALER_DIV1;
  huart4.AdvancedInit.AdvFeatureInit = UART_ADVFEATURE_NO_INIT;

  if (HAL_UART_Init(&huart4) != HAL_OK)
  { 
    Serial.println("\nUart4 Error");
  }
  /* USER CODE BEGIN USART4_Init 2 */
  
  if (HAL_UARTEx_SetTxFifoThreshold(&huart4, UART_TXFIFO_THRESHOLD_1_8) != HAL_OK)
  {
    Serial.println("\nUart4 Error");
  }

  if (HAL_UARTEx_SetRxFifoThreshold(&huart4, UART_RXFIFO_THRESHOLD_1_8) != HAL_OK)
  {
    Serial.println("\nUart4 Error");
  }

  if (HAL_UARTEx_DisableFifoMode(&huart4) != HAL_OK)
  {
    Serial.println("\nUart4 Error");
  } 
}

```

fpistm, 2020-09-24:
Hi @harji2130 
I found the issue.
After fixing this issue https://github.com/stm32duino/Arduino_Core_STM32/pull/1141

I forgot to update the IRQ handler function then for USART4 the  `HAL_UART_IRQHandler` is not called.
I will fix it too...

matthijskooijman, 2021-04-01:
As a workaround, you can now explicitly *disable* remapping again by using PA11 and PA12 explicitly. e.g.:

```C++
HardwareSerial rfUart(PA10, PA9);

void setup() {
  rfUart.begin(9600);
  pinMode(PA11, INPUT);
  pinMode(PA12, INPUT);
}
```

PR #1091 (2021-04-17): [variant] Improve variant support and genericity
As the number of variants continues to grow and in order to ease navigation, they have been grouped in sub-folders per 
series.

To improve variants support and generic ones several enhancements/fixes has been done:
- Each variant will be linked to its mcu/package
- All pins capabilities will be possible without modifying the `PeriperhalPins.c` PinMap array. ALTx pin naming will allow to use all alternative possibilities which use other HW peripheral instances.
- All remap pins will be supported (F0/G0)
- All dual pad pins will be supported (H7) [... cut]
commit 0898a854 (2021-04-17): Remap pin management

Fix #1180
closed 2021-04-17 by fpistm