stm32duino/Arduino_Core_STM32 #1183: TwoWire requestFrom(uint8_t address, size_t size, bool sendStop) Implementation

porkyneal, 2020-09-22:
I am having a Platformio build error with a library [(ArduinoECCX08)](https://github.com/arduino-libraries/ArduinoECCX08).

This is because it calls:

```while (_wire->requestFrom((uint8_t)_address, (size_t)responseSize, (bool)true) != responseSize && retries--);```

Where it s casting the sendStop paramater as a boolean, the compiler fails saying that it is ambiguous. The core does not appear to have the Arduino specific function as found here --> [Link](https://github.com/esp8266/Arduino/blob/master/libraries/Wire/Wire.cpp#L123)

**Describe the solution you'd like**
Implement a handler that better aligns with the arduino specfification --> [Docs](https://www.arduino.cc/en/Reference/WireRequestFrom)

Are there any guides I need to follow to implement a PR for this repo?

fpistm, 2020-09-22:
Hi @porkyneal 
Thanks for the PR.
This required some check. Looking at several Wire implementation in different core only the samd and megaavr have this one , that's why the library uses this prototyp as it is only compatible for those core.
Is the library works fine with your PR ?

porkyneal, 2020-09-22:
Hi @fpistm, I wasn't sure if it was worth a PR as obviously it is no bug, just an issue with this particular library casting to that exact defintion of requestFrom().

I guess if you are a stickler to align to the exact Arduino spec then this it is required 🤷 ?

Yes, my project compiles and works with this change in Platformio.

fpistm, 2020-09-22:
We will see how to answer this properly, maybe your PR is the best choice anyway I would like investigate to ensure no backward compatibility and also best integration.

PR #1184 (2020-10-20): Add requestFrom(uint8_t address, size_t quantity, bool sendStop)
This PR adds a missing function declaration that aligns further with the arduino spec as specified in issue #1183 

This is to fix compilation with an external Arduino library [ArduinoECCX08](https://github.com/arduino-libraries/ArduinoECCX08)

Fixes #1183
closed 2020-10-20 by ABOSTM