stm32duino/Arduino_Core_STM32 #1193: [CDC] Serial USB without DTR

jlocsx, 2020-09-25:
**Is your feature request/improvement related to a problem? Please describe.**

I recently experienced the following issue using the [Blue Pill](https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill.html) board ([STM32F103C8T6](https://www.st.com/en/microcontrollers-microprocessors/stm32f103c8.html) MCU) with the STM32Duino core:

- Using the USB CDC, the data sent by Blue Pill was not received by an old Terminal Serial written via Delphi 7;
- For example, this Terminal was unable to receive data from a program based on the following code:
    - https://gist.github.com/jlocsx/f01631e6109fff5e485fbd998b28a8cf

This is because the DTR signal is not enabled by default on that Serial Terminal. After enabling this signal, I was able to receive the data without any problem.

But I didn't experience any issue when I performed the tests using a [Rust](https://www.rust-lang.org/what/embedded) program based on the following code with the Blue Pill:
- https://github.com/stm32-rs/stm32-usbd-examples/blob/master/example-stm32f103c8/src/main.rs

I also tested a Honeywell RFID reader ([YJ-HF500](https://country.honeywellaidc.com/CatalogDocuments/YJ-HF500H-EN-UG.pdf)) that communicate via USB CDC, and the Terminal was able to receive the data even with the DTR signal disabled.

**Describe the solution you'd like**

I found the following "work around" in the STM32duino forum:
- https://www.stm32duino.com/viewtopic.php?f=62&t=461

But I would like something that can be configured through the development environment, without having to modify the `usbd_cdc_if.c` file directly.

For example, I would like to be able to enable/ disable the DTR signal verification by overwriting some macro. Or is this already possible?

fpistm, 2020-09-26:
Hi @jlocsx 
do not hesitate to provide a PR for this as mentioned in the forum any contribution are welcome. 😉

maxgerhardt, 2021-03-17:
This issue just hit me (and [other guy](https://community.platformio.org/t/cannot-upload-to-stm32-blue-pill-fake-after-pio-update/19415/4?u=maxgerhardt)), too. I was using [hterm](http://der-hammer.info/pages/terminal.html) and `miniterm.py` (serial monitor in PlatformIO) as my serial monitor programs and saw no output -- believing that something was seriously broken. Then I found this issue and activated the DTR line, and in that moment output came through again. 

What was the rationale behind the change? Per above two widely used serial monitors don't set the DTR flow control lines upon opening the virtual COM port, leading to this strange behavior.

ag88, 2021-04-09:
hi for those using Linux there is one thing to check 
- create a udev rules file e.g.
/etc/udev/rules.d/45-stm32duino.rules
add a line like
```
ATTRS{idProduct}=="5740", ATTRS{idVendor}=="0483", MODE="664", GROUP="plugdev" SYMLINK+="stm32duino" ENV{ID_MM_DEVICE_IGNORE}="1"
```
make sure that the product id and vendor id matches that used in your sketch.

the main thing is that 
``
ENV{ID_MM_DEVICE_IGNORE}="1"
``
which tells udev not to treat it as a modem
that issue is explored in this thread
https://www.stm32duino.com/viewtopic.php?f=62&t=1000

fpistm, 2021-04-09:
>  which tells udev not to treat it as a modem

I though it was fixed by https://github.com/stm32duino/Arduino_Core_STM32/pull/884

ag88, 2021-04-09:
hi fpistm, i'd answer in the other thread

fpistm, 2021-05-04:
Hi @jlocsx, @maxgerhardt and @ag88 

I've made a PR to implement a way to disable the DTR usage. See #1382.
If you can test and/or have any feedback this will be welcome.

I've tested using hterm as stated by @maxgerhardt and it works as expected.

Hope it is fine for you. 😉

closed 2021-05-05 by fpistm

ivan-habl, 2022-08-06:
Now, in order to observe UART data over USB, you must either use a terminal program with DRT monitoring support or force SerialUSB.dtr(true) into loop().

We need to make this optional, not the default behavior.

fpistm, 2022-08-23:
By default DTR is enabled as before. This PR simply implement a way to disable it. If you don't need it simply call `SerialUSB.dtr(false)`

PR #1382 (2021-05-05): USB CDC line state
Add api to manage the DTR usage.
 Usage: 
```C++
Serial.dtr(false); // Disable the DTR
Serial.dtr(true); // Enable the DTR - default state
```

Assuming `Serial` is mapped on `SerialUSB`

Also implement `rts()` and `dtr()` to return correct state,

Fixes #1193
commit f04c66b1 (2021-05-05): USB: CDC: add api to manage the DTR usage

Fixes #1193