stm32duino/Arduino_Core_STM32 #1195: 3 UART ports in Generic STM32F4 - Generic F407VG

maks-davletchin, 2020-09-29:
Hi.
I chose Generic STM32F4 - Generic F407VG board in Arduino IDE.
In folder ..\STM32\hardware\stm32\1.9.0\variants\Generic_F4x7Vx\ I found 5 files and open variant.h
There are the next strings:

`// UART Definitions
#define SERIAL_UART_INSTANCE    2

// Default pin used for 'Serial' instance
// Mandatory for Firmata
#define PIN_SERIAL_RX           PA3
#define PIN_SERIAL_TX           PA2`

But in this case, when i use `Serial.println("text")` there are no any data in pins PA3 and PA2 (there are in Serial2), but I realized that Serial is for USB and Serial2 is for USART2 (PA3 and PA2).

For the experiment I changed  `PIN_SERIAL_RX` to PA10 and `PIN_SERIAL_TX` to PA11 (USART1), saved and upload sketch to mc. But this changes didn't apply. And Serial2 continues to use pins USART2 (PA3 and PA2).

The first question:
Why can't I change pins to choose another USART?

Now I'd like to use 3 UART ports in my board (DevEBox from Aliexpress).

In file variant.h I change and add some strings:

`// UART Definitions
#define SERIAL_UART_INSTANCE    3

// Default pin used for 'Serial' instance
// Mandatory for Firmata
// It must be Serial1
#define PIN_SERIAL_RX           PA3   //USART2_RX
#define PIN_SERIAL_TX           PA2   //USART2_TX

//  It must be Serial2
#define PIN_SERIAL_RX1         PA10  //USART1_RX
#define PIN_SERIAL_TX1         PA9    //USART1_TX

//  It must be Serial3
#define PIN_SERIAL_RX2         PB11  //USART3_RX
#define PIN_SERIAL_TX2         PB10  //USART3_TX
`

```
#define SERIAL_PORT_HARDWARE1             Serial1
#define SERIAL_PORT_HARDWARE_OPEN1  Serial1
#define SERIAL_PORT_HARDWARE2             Serial2
#define SERIAL_PORT_HARDWARE_OPEN2  Serial2
#define SERIAL_PORT_HARDWARE3             Serial3
#define SERIAL_PORT_HARDWARE_OPEN3  Serial3
```

The second question:
What files should I change else so my ports will be working?

fpistm, 2020-09-29:
Hi @maks-davletchin

> What files should I change else so my ports will be working?

None. You should not have to change the core file.

> But in this case, when i use `Serial.println("text")` there are no any data in pins PA3 and PA2 (there are in Serial2), but I realized that Serial is for USB and Serial2 is for USART2 (PA3 and PA2).

`Serial `is simply a define to the default Serialx used. In this case, if USB CDC is not enabled, `Serial `is `Serial2 `else `Serial `is `SerialUSB`.

See: https://github.com/stm32duino/wiki/wiki/API#hardwareserial

To add extra `PIN_SERIAL_RX/TXn` definition you can use the (hal_conf_extra.h)[https://github.com/stm32duino/wiki/wiki/HAL-configuration#customize-hal-or-variant-definition] or (build_opt)[https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h]

Please for this kind of question/support use the forum: http://www.stm32duino.com

maks-davletchin, 2020-09-29:
@fpistm , thank you for your fast answer!
Will try

> Please for this kind of question/support use the forum: http://www.stm32duino.com

Ok, no problems

closed 2020-10-06 by fpistm