stm32duino/Arduino_Core_STM32 #1197: In stm32f103c8t6, GPIOA1, 2, 3 and GPIOA15 cannot be used together with pwm

yulong88888, 2020-10-06:
I am very excited when I use stm32duino for the first time. Thank you very much for your contribution.

I need to use GPIOA1, 2, 3 and GPIO15 to output PWM. When using the analogWrite function, two things will happen. First, 1 2 3 can output pwm, 15 can't, then 15 can output pwm, 1 2 3 can’t. I am using stm32f103c8t6.

I will also check the chip manual, but I still hope to get help, thank you very much.

[image]

fpistm, 2020-10-07:
Hi @yulong88888 
I confirmed your issue.
We guess it is linked to the PA15 GPIO config which has a special config as it is a JTAG pins:
`__HAL_AFIO_REMAP_SWJ_NOJTAG(); // JTAG-DP Disabled and SW-DP enabled`

Using PA0 which relies on the same Timer instance and channel than PA15 (TIM2_CH1) it works fine with the 4 pins.

Sample sketch to reproduce:
```
// the setup routine runs once when you press reset:
void setup() {
  // declare pins to be an output:
  //pinMode(PA0, OUTPUT);
  pinMode(PA1, OUTPUT);
  pinMode(PA2, OUTPUT);
  pinMode(PA3, OUTPUT);
  pinMode(PA15, OUTPUT);
}

// the loop routine runs over and over again forever:
void loop() {
  static uint32_t brightness = 0;
  static uint32_t fadeAmount = 5;
  // set the brightness of pin:

  //analogWrite(PA0, brightness);
  analogWrite(PA1, brightness);
  analogWrite(PA2, brightness);
  analogWrite(PA3, brightness);
  analogWrite(PA15, brightness);

  // change the brightness for next time through the loop:
  brightness = brightness + fadeAmount;

  // reverse the direction of the fading at the ends of the fade:
  if (brightness <= 0 || brightness >= 255) {
    fadeAmount = -fadeAmount;
  }
  // wait for 30 milliseconds to see the dimming effect
  delay(30);
}
```

ABOSTM, 2020-10-07:
Hi @yulong88888,
Problem is linked to the hardware remap capabilities of STM32F1xx (see reference manual: RM0008) :
[image]
There is no TIM2_REMAP value that match  PA15 (TIM2_CH1) and PA1 (TIM2_CH2) at the same time.

fpistm, 2020-10-07:
Thanks @ABOSTM 

@yulong88888, note that you can customize the PinMap_PWM as explained here but ensure to be coherent on your pin mapping and the alternate function:
https://github.com/stm32duino/wiki/wiki/Custom-definitions#custom-pinmap-array

closed 2020-10-07 by fpistm

yulong88888, 2020-10-07:
Thanks, everyone!