stm32duino/Arduino_Core_STM32 #1199: When using both USB Serial and Serial1 on stm32f103 boards port pins for serial1 are not configured

J-f-Jensen, 2020-10-08:
I have after a bunch of new gray hairs found out why my GPS code not worked as expected when I used the USB serial port for PC communication and Serial1 for GPS communication.
Pin definition for Serial1 is missing from variant.h, after I have added the following two lines it works both with Serial via uart and with Serial (USB) and Serial1 enabled
#define PIN_SERIAL1_RX           PA10
#define PIN_SERIAL1_TX           PA9

This seams to be a issue for most variants where both Serial via USB and uart can be enabled at the same time

fpistm, 2020-10-08:
If you have this config, it works as expected:
[image]

Only `Serial `is remapped to `SerialUSB `instead of `Serial1` but Seria1 is instantiate by default.
If you used "Enabled (No generic Serial)", in that case Serial1 is not instantiate.

https://github.com/stm32duino/wiki/wiki/API#hardwareserial

J-f-Jensen, 2020-10-09:
Wird! I use exact the same configuration. I had more than one arduino sketch open with different configurations, after cleanup and restart it works as expected. Thanks for the fast response

closed 2020-10-09 by J-f-Jensen