stm32duino/Arduino_Core_STM32 #1206: Nucleo STM32F401RE: Program received signal SIGTRAP, Trace/breakpoint trap.

rei-vilo, 2020-10-13:
**Describe the bug**

Debugging the `blink` example using release 1.8.0 raises the error.

> Program received signal SIGTRAP, Trace/breakpoint trap.

I'm using release 1.8.0 as executable built with release 1.9.0 doesn't run. See #1205 

**To Reproduce**

Steps to reproduce the behavior:

1. Launch Arduino IDE
2. Select Blink example
3. Select following options

[image]

4. Export compiled binary

5. Open first Terminal window
6. Launch `st-util -p 3333`

7. Open second Terminal window
8. Launch `/Users/USER/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-gdb`

9. On the GDB console, run

```
GNU gdb (xPack GNU Arm Embedded GCC, 64-bit) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin14.5.0 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file /var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino_build_931168/Blink.ino.elf
Reading symbols from /var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino_build_931168/Blink.ino.elf...
(gdb) set remotetimeout 2000
(gdb) target extended :3333
Remote debugging using :3333
0x61000000 in ?? ()
(gdb) monitor reset halt
(gdb) load
Loading section .isr_vector, size 0x194 lma 0x8000000
Loading section .text, size 0x4f2c lma 0x8000194
Loading section .rodata, size 0x510 lma 0x80050c0
Loading section .ARM, size 0x8 lma 0x80055d0
Loading section .init_array, size 0xc lma 0x80055d8
Loading section .fini_array, size 0x8 lma 0x80055e4
Loading section .data, size 0x20c lma 0x80055ec
Start address 0x8003234, load size 22520
Transfer rate: 16 KB/sec, 2815 bytes/write.
(gdb) monitor reset init
(gdb) list loop()
27	  // initialize digital pin LED_BUILTIN as an output.
28	  pinMode(LED_BUILTIN, OUTPUT);
29	}
30	
31	// the loop function runs over and over again forever
32	void loop() {
33	  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
34	  delay(1000);                       // wait for a second
35	  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
36	  delay(1000);                       // wait for a second
(gdb) break 33
Breakpoint 1 at 0x8000216: file /Applications/IDE/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino, line 33.
(gdb) enable 1
(gdb) c
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Program received signal SIGTRAP, Trace/breakpoint trap.
main ()
    at /Users/USER/Library/Arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/main.cpp:65
65	    if (serialEventRun) {
(gdb) frame 0
#0  main ()
    at /Users/USER/Library/Arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/main.cpp:65
65	    if (serialEventRun) {
(gdb) info frame 0
Stack frame at 0xa037a08:
 pc = 0x80036b4 in main
    (/Users/USER/Library/Arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/main.cpp:65); saved pc = <not saved>
 Outermost frame: Cannot access memory at address 0xa037a04
 source language c++.
 Arglist at 0xa037a00, args: 
 Locals at 0xa037a00, Previous frame's sp is 0xa037a08
Cannot access memory at address 0xa037a00
(gdb) 
```

**Expected behavior**

No SIGTRAP

**Screenshots**

See above

- OS: macOS 10.15.7
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.8.0
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial] See above
 - Upload method: [e.g. STLink] MSD

**Board (please complete the following information):**

 - Name: Nucleo F401RE
 - Hardware Revision: MB1136 rec.C
 - Extra hardware used if any: none

**Additional context**

Thank you!

rei-vilo, 2020-10-13:
Adding the `extern "C" void SystemClock_Config(void)` function did the trick with release 1.9.0.

Also, I had to change `Nucleo_64.menu.pnum.NUCLEO_F401RE.node=NODE_F401RE` for `Nucleo_64.menu.pnum.NUCLEO_F401RE.node=NUCLEO` in order to upload, as the board shows up as `NUCLEO` and not `NODE_F401RE`.

Next step: testing debugging.

fpistm, 2020-10-13:
About the node name, it is a known issue, old board have this naming. Updating the STLink fw will solve this issue, I think.
Else, the name "NUCLEO" can be added to the node name in boards.txt:
https://github.com/stm32duino/wiki/wiki/FAQ#when-using-mass-storage-upload-method-following-error-is-displayed

For the clock issue, it seems your board have an issue with the HSE Bypass  (clock provided by the STLink).

For the debugging I will not be able to help. Anyway I'm wondering as it seems you have a STLink clock issue then debug could be impacted.

rei-vilo, 2020-10-13:
I performed an update of the ST-Link V.2.1 firmware, to no avail.

[image]

The board might be too old, with serial number `MB1136 rev.C`. 

I'll try with other Discovery boards I have at hand, STM32F429I and STM32F746G.

So I'm closing the thread.