stm32duino/Arduino_Core_STM32 #1207: TwoWire endTransmission(uint8 sendStop) Implementation

trawinlegacy, 2020-10-14:
**Is your feature request/improvement related to a problem? Please describe.**
The STM32F4 boards cannot operate with all Arduino based I2C sensor libraries. Ex: MPL3115A

When the endTransmission(false) is removed from the library function call the STM32 does not pull the right device address.

**Describe the solution you'd like**
A endTransmission(uint8 sendStop) with backwards compatibility for endTransmission(void).

**Describe alternatives you've considered**
I adjusted the Wire.cpp to include the following:
`
uint8 TwoWire::endTransmission(uint8_t sendStop) { // ******* originally - endTransmission(void) *******
if (tx_buf_overflow) return EDATA;

i2c_start(port);

i2c_shift_out(port, (tx_addr << 1) | I2C_WRITE);
if (!i2c_get_ack(port)) return ENACKADDR;

// shift out the address we're transmitting to
for (uint8 i = 0; i < tx_buf_idx; i++) {
	uint8 ret = writeOneByte(tx_buf[i]);
	if (ret) {
		//Serial1.println("endTransmission failed");
		return ret;    // SUCCESS is 0
	}
}

if (sendStop) {				// ******* originally - i2c_stop(port); with no if statement *******
	i2c_stop(port);
}

tx_buf_idx = 0;
tx_buf_overflow = false;
return SUCCESS;
}

// This provides backwards compatibility with the original
// definition, and expected behaviour, of endTransmission
//
uint8 TwoWire::endTransmission(void) // ******* Originally function was not included *******
{
return endTransmission(true);
}
`

I also adjusted the Wire.h to include the following
`uint8 endTransmission(void); 
uint8 endTransmission(uint8_t);`

This allows the code to run but it does not read the sensor properly as these are the results from the MPL3115A2 sensor connected:

-11.25*C

When it should be 22.5*C

fpistm, 2020-10-14:
Hi @trawinlegacy 

I don't understand your point as the API is already available:
https://github.com/stm32duino/Arduino_Core_STM32/blob/cbccfe4938f867e1025807d1b05ff8260224485e/libraries/Wire/src/Wire.h#L95-L96

https://github.com/stm32duino/Arduino_Core_STM32/blob/cbccfe4938f867e1025807d1b05ff8260224485e/libraries/Wire/src/Wire.cpp#L206-L276

Moreover I didn't recognize some of the function you used like : `i2c_start`, `i2c_shift_out`, `writeOneByte` so I guess you do not use this core.

fpistm, 2020-10-14:
I've found the related core:
https://github.com/Seeed-Studio/stm32f4ArduinoCore/blob/master/libraries/Wire/Wire.h

So please open issue to this core.

closed 2020-10-14 by fpistm

trawinlegacy, 2020-10-14:
Wow, so sorry. Don't know how I missed that.... I swear I didn't see it in the file.