stm32duino/Arduino_Core_STM32 #1210: Problem With Output pwm At Pa0 & Pa8

aghfars, 2020-10-20:
hello i want to have a Separately PWM at pin PA0 & PA8 but when i set PA8 on 127 or any number on (8 bit Resolution) i have the same output at pin PA0 & And conversely.
iam not test on another Resolution. I'm in a hurry 
Help..

fpistm, 2020-10-20:
Hi without more details, we could not help.
Which board you used? Core version?...

Pins are probably using the same timer instance and channel...

aghfars, 2020-10-20:
> Hi without more details, we could not help.
> Which board you used? Core version?...
> 
> Pins are probably using the same timer instance and channel...

the board is stm32f103c8t6 Bluepill (64kB)
core Version is 1.9.0
no , one pin on timer 1 and another one on timer 2
and i test again when i write my servo pos on pin PA8 and after that i call the analogWrite on pin PA0 or PA3 have same pwm value on PA8 and PA0 or PA3. (Makes the servo motor rotate 360 degrees like a normal motor)
i didnt test with PA2 & PA3.
the is:
Now what I have done to solve the problem is that before giving the angle to the servo, I attach the servo and after a while I detach it and write the analogWrite again.

ABOSTM, 2020-10-20:
Hi @aghfars,
Looking at bluepill PeripheralPins.c
https://github.com/stm32duino/Arduino_Core_STM32/blob/2d652e239825c639160a5110690cfaf2c940e385/variants/PILL_F103XX/PeripheralPins.c#L98
PA0 is mapped on TIM2

and looking at variant.h
https://github.com/stm32duino/Arduino_Core_STM32/blob/2d652e239825c639160a5110690cfaf2c940e385/variants/PILL_F103XX/variant.h#L112
Servo is mapped also on TIM2

This is clearly incompatible. If possible I suggest you to use TIM6 or TIM7 for Servo(if not used in your application), because they have no hardware output which is fine for servo.

aghfars, 2020-10-20:
> Hi @aghfars,
> Looking at bluepill PeripheralPins.c
> https://github.com/stm32duino/Arduino_Core_STM32/blob/2d652e239825c639160a5110690cfaf2c940e385/variants/PILL_F103XX/PeripheralPins.c#L98
>
>
> PA0 is [...]

thanks for I did not use timers 6 and 7 unless used in libraries.

closed 2020-10-30 by fpistm