stm32duino/Arduino_Core_STM32 #1211: Not found STM32 Cores by STMicroelectronics on Library Manger of Arduino IDE

FaustoGiudice, 2020-10-20:
I followed the instructions to install STM32 Cores by STMicroelectroni. I have set the link for JSON file, but the manager libraries cannot find STM32 Cores by STMicroelectronics. I have tried on several computers, nothing to do!

I follow this indication 
https://www.maffucci.it/2017/12/07/stm-32-nucleo-supporto-arduino-core/ 

strangely I can't even find ESP8266 library!

my settings word in the Preference menu -> Additional Board Manager URLs :

http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

fpistm, 2020-10-20:
>  manager libraries

This is not a library, this is a core. You have to use the Boards manager.

FaustoGiudice, 2020-10-21:
Also Board Manager not found that Core and never ESP8266  :-(

fpistm, 2020-10-21:
which version of Arduino IDE  you used ?
Any error displayed?

I guess the package index json was corrupted. You wcould try to delete them manually.

FaustoGiudice, 2020-10-21:
I solved my Problem!  Thank ;-)

closed 2020-10-21 by fpistm