stm32duino/Arduino_Core_STM32 #1213: SparkFun_RFD77402_Arduino_Library to much timeout

aghfars, 2020-10-20:
i have problem with this library and cjmcu-402
i have to much time out and errors when i want to get distance.
my core ver is : 1.9.0
and stm32f103c8t6 bluepill(64K) is my board
and i use tim4 and tim2 on my code
i havnt any problem at mega2560

fpistm, 2020-10-21:
Hi @aghfars 

Did you wired pull up resistor on each I2C line ? (it seems they are available on the shield anyway it can be disconnected)
As it is a third party library I could not help on this as I don't have the shield and the product is retired.

You should open a topic on this on the forum: https://www.stm32duino.com

closed 2020-10-21 by fpistm

aghfars, 2020-10-21:
> Hi @aghfars
>
> Did you wired pull up resistor on each I2C line ? (it seems they are available on the shield anyway it can be disconnected)
> As it is a third party library I could not help on this as I [...]

> Hi @aghfars
>
> Did you wired pull up resistor on each I2C line ? (it seems they are available on the shield anyway it can be disconnected)
> As it is a third party library I could not help on this as I [...]

yes i have pull up resistor
i test it with mega2560 the reading sensor between 50 to 53 has delta time.
but with stm32f103c8t6 i have only two delta time With two values of 50 and 60 ms.
i think mega2560 is More reliable than stm32f103c8t6.
delay used in lib and i think the problem about it or millis function has problem.