stm32duino/Arduino_Core_STM32 #1214: Arduino_Core_STM32/platform.txt needs tweek to make CMSIS fast math

Duffmann, 2020-10-20:


Duffmann, 2020-10-20:
CMSIS DSP math functions like `arm_sin_f32()` from *static* lib `libarm_cortexM4lf_math.a` fail to link properly in Arduino

...
c:/users/x/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: ESC_PWM_Commuting.ino.cpp:(.text.loop+0x84): **undefined reference to `arm_sin_f32'**
collect2.exe: error: ld returned 1 exit status

**Fix**
Move `{compiler.arm.cmsis.ldflags} ` between `-Wl,--start-group` / `-Wl,--end-group` statements in line `recipe.c.combine.pattern=` inside platform.txt

**Current  recipe.c.combine.pattern in platform.txt**
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} "-Wl,--default-script={build.variant.path}/{build.ldscript}" "-Wl,--script={build.system.path}/ldscript.ld" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} {compiler.ldflags} **{compiler.arm.cmsis.ldflags}** -o "{build.path}/{build.project_name}.elf" "-L{build.path}" **-Wl,--start-group** {object_files} {compiler.libraries.ldflags} "{archive_file_path}" -lc -**Wl,--end-group** -lm -lgcc -lstdc++

**Fixed recipe.c.combine.pattern in platform.txt**
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} "-Wl,--default-script={build.variant.path}/{build.ldscript}" "-Wl,--script={build.system.path}/ldscript.ld" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} {compiler.ldflags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -**Wl,--start-group** {object_files} {compiler.libraries.ldflags} "{archive_file_path}" **{compiler.arm.cmsis.ldflags}** -lc -**Wl,--end-group** -lm -lgcc -lstdc++

Found the reason for this issue and the solution here [here ](https://stackoverflow.com/questions/18505221/significance-of-start-group-and-end-group-linker-flags-for-dynamic-libraries)

fpistm, 2020-10-20:
Hi @Duffmann 
You have to use the CMSIS DSP built in library: https://github.com/stm32duino/wiki/wiki/API#cmsis-dsp

I've tested this [example](https://www.keil.com/pack/doc/CMSIS/DSP/html/arm_sin_cos_example_f32_8c-example.html) converted to Arduino sketch and it works fine :
```C++
/* ----------------------------------------------------------------------
  Copyright (C) 2010-2012 ARM Limited. All rights reserved.

  $Date:         12. March 2014
  $Revision:     V1.4.3

  Project:       CMSIS DSP Library
  Title:         arm_sin_cos_example_f32.c

  Description:   Example code demonstrating sin and cos calculation of input signal.

  Target Processor: Cortex-M4/Cortex-M3

  Redistribution and use in source and binary forms, with or without
  modification, are permitted provided that the following conditions
  are met:
    - Redistributions of source code must retain the above copyright
      notice, this list of conditions and the following disclaimer.
    - Redistributions in binary form must reproduce the above copyright
      notice, this list of conditions and the following disclaimer in
      the documentation and/or other materials provided with the
      distribution.
    - Neither the name of ARM LIMITED nor the names of its contributors
      may be used to endorse or promote products derived from this
      software without specific prior written permission.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
  "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
  LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
  FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
  COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
  INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
  BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
  LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
  ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
  POSSIBILITY OF SUCH DAMAGE.
  -------------------------------------------------------------------- */
#include <CMSIS_DSP.h>

/* ----------------------------------------------------------------------
  Defines each of the tests performed
  ------------------------------------------------------------------- */
#define MAX_BLOCKSIZE   32
#define DELTA           (0.0001f)
/* ----------------------------------------------------------------------
  Test input data for Floating point sin_cos example for 32-blockSize
  Generated by the MATLAB randn() function
  ------------------------------------------------------------------- */
const float32_t testInput_f32[MAX_BLOCKSIZE] =
{
  -1.244916875853235400,  -4.793533929171324800,   0.360705030233248850,   0.827929644170887320,  -3.299532218312426900,   3.427441903227623800,   3.422401784294607700,  -0.108308165334010680,
  0.941943896490312180,   0.502609575000365850,  -0.537345278736373500,   2.088817392965764500,  -1.693168684143455700,   6.283185307179590700,  -0.392545884746175080,   0.327893095115825040,
  3.070147440456292300,   0.170611405884662230,  -0.275275082396073010,  -2.395492805446796300,   0.847311163536506600,  -3.845517018083148800,   2.055818378415868300,   4.672594161978930800,
  -1.990923030266425800,   2.469305197656249500,   3.609002606064021000,  -4.586736582331667500,  -4.147080139136136300,   1.643756718868359500,  -1.150866392366494800,   1.985805026477433800
};
const float32_t testRefOutput_f32 = 1.000000000;
/* ----------------------------------------------------------------------
  Declare Global variables
  ------------------------------------------------------------------- */
uint32_t blockSize = 32;
float32_t  testOutput;
float32_t  cosOutput;
float32_t  sinOutput;
float32_t  cosSquareOutput;
float32_t  sinSquareOutput;
/* ----------------------------------------------------------------------
  Max magnitude FFT Bin test
  ------------------------------------------------------------------- */
arm_status status;
void setup() {
  float32_t diff;
  uint32_t i;

  Serial.begin(9600);
  while(!Serial);
  for (i = 0; i < blockSize; i++) {
    cosOutput = arm_cos_f32(testInput_f32[i]);
    sinOutput = arm_sin_f32(testInput_f32[i]);
    arm_mult_f32(&cosOutput, &cosOutput, &cosSquareOutput, 1);
    arm_mult_f32(&sinOutput, &sinOutput, &sinSquareOutput, 1);
    arm_add_f32(&cosSquareOutput, &sinSquareOutput, &testOutput, 1);
    /* absolute value of difference between ref and test */
    diff = fabsf(testRefOutput_f32 - testOutput);
    /* Comparison of sin_cos value with reference */
    status = (diff > DELTA) ? ARM_MATH_TEST_FAILURE : ARM_MATH_SUCCESS;
    if ( status == ARM_MATH_TEST_FAILURE) {
      break;
    }
  }
  if (status != ARM_MATH_SUCCESS) {
    Serial.printf("FAILURE\n");
  } else {
    Serial.printf("SUCCESS\n");
    while (1);                             /* main function does not return */
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Duffmann, 2020-10-20:
`#include <CMSIS_DSP.h>` works like a charm with the untouched platform.txt!

My fault - I previously only added
```
#define ARM_MATH_CM4
#include <arm_math.h>
```
Thanks for the super fast fix

closed 2020-10-20 by Duffmann