stm32duino/Arduino_Core_STM32 #1216: 32F767ZI pin definitions missing

TadyTheFish, 2020-10-23:
Hello
I'm trying to use some pins on the 767ZI and It seems some pins are not defined. Is there a special reason for this? 
And what is the logic behind the numbering. It seems it does not follow the psychical pin numbering.

Thank you

fpistm, 2020-10-23:
Hi @TadyTheFish 
it is possible that not all pins were not be defined.
In a general way the pin numbering follow the Arduino Connector the the ST morpho connector is added.
I'm currently rework the variants management to be more generic and provide access to all peripheral per pins without change the PinMap array.

TadyTheFish, 2020-10-23:
Ok thank you.. Can I do anything  so I could test a project I'm working on?

fpistm, 2020-10-23:
You can modify the variant to add the missing pins.
Which ones are missing ?

TadyTheFish, 2020-10-23:
I don't know. For now I can say only for PA11,PA12,PC5 .. They are located on CN12 on the NUCLEO-F767ZI board. I think the black connectors are define but the empty vias (CN12 and CN11) are not.

TadyTheFish, 2020-10-23:
also PA1 ,PA12 ,PA11, PH0 PH1 ,PE1, PG12,  PH2, PG11, PG13, PG10,PG15,PC1, PA9,PA8,PA10, PA2,PD10,PG7,PG4, PC4,PG8,PG5,PG6

Tese are on the CN11 and CN12 connectors.. I didn't go trough the black connectors

fpistm, 2020-10-23:
OK. You can add them to the variant.[h|cpp] and update the number of NUM_DIGITAL_PINS.

TadyTheFish, 2020-10-23:
Would love to but I can't figure out how the numbering system work.. there is no order.

fpistm, 2020-10-23:
This is just a number important things is the pin number defined in the variant.h is the index of the PinName in the digital pins array.

TadyTheFish, 2020-10-23:
I see pinNames have underscores. But I still don't see how they correlate to pin numbers on the datasheet. There must be a link.

TadyTheFish, 2020-10-23:
Ok pinNames.h holds some info... Defines of ports and offsets .. I will try it

TadyTheFish, 2020-10-23:
Interesting .. Now I probably know why those pines are not defined. I had a problem with the USB connection. When looking trough the schematic I fond out that for example PA8 ,PA9,PA10,PA11,PA12 are used for the USB :D

fpistm, 2020-10-30:
Yes some pins have a dedicated usage. Anyway if you do not enable USB you can use it for other thing except if some components avoid this (PU/PD resistor, capacitor,...)
I close this issue as I considered it as answered. Other pins will probably be added in variant rework.

closed 2020-10-30 by fpistm