stm32duino/Arduino_Core_STM32 #1218: How to change cpu clock while mcu is running?

thebao1992, 2020-10-24:
I have a project using stm32f407 to control Oled SSD1306 (show built-in rtc clock, oled "always on display"), 16*WS2812 * 6, lots of DS18B20, ILI9488, GPS, MPU6050,, Hall sensor, relay,...
my code is over 7000 line so i can't post it here.
Change cpu speed by edit Variant.cpp is ok but i want to slow cpu clock while mcu is running, mean sketch is still running. Can you help me?
Ex: 
void setup() {
pinMode(PA0, INPUT_PULLDOWN);
}
void loop() {
if (digitalRead(PA0) == HIGH) {
// do something to run normal speed
}
if (digitalRead(PA0) == LOW) {
// do something to run very slow speed
}
}

fpistm, 2020-10-25:
Hi @thebao1992 
You do not need to change the variant.cpp.
The `SystemClock_Config` is a weak function so redefine it in your sketch prefixing by `extern "C"`.
```c
extern "C" void SystemClock_Config(void) {
...
}
```
Then you can define an other function for your low power clock config anyway pays attention to update all used peripheral to be inline with the new clock config. For example if you use an `HardwareSerial ` it will required to reinit them to take in account the change.

Please avoid to open an issue for requesting support. Use the forum wxhich is dedicated to this: https://www.stm32duino.com

closed 2020-10-25 by fpistm

thebao1992, 2020-10-30:
extern "C" void SystemClock_Config(void) { }

void setup() {
pinMode(PA0, INPUT_PULLDOWN);
}
void loop() {
SystemClock_Config(void)
{
  RCC_ClkInitTypeDef RCC_ClkInitStruct;
  RCC_OscInitTypeDef RCC_OscInitStruct;

  /* Enable Power Control clock */
  __HAL_RCC_PWR_CLK_ENABLE();

  /* The voltage scaling allows optimizing the power consumption when the device is
     clocked below the maximum system frequency, to update the voltage scaling value
     regarding system frequency refer to product datasheet.  */
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

  /* Enable HSE Oscillator and activate PLL with HSE as source */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 8;

if (digitalRead(PA0) == LOW) {
  RCC_OscInitStruct.PLL.PLLN = 200;  //Slow speed cpu
}

if (digitalRead(PA0) == HIGH) {
RCC_OscInitStruct.PLL.PLLN = 336;  //Normal speed
}

  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  HAL_RCC_OscConfig(&RCC_OscInitStruct);

  /* Select PLL as system clock source and configure the HCLK, PCLK1 and PCLK2
     clocks dividers */
  RCC_ClkInitStruct.ClockType = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK |
                                 RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2);
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;
  HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5);

  /* STM32F405x/407x/415x/417x Revision Z devices: prefetch is supported  */
  if (HAL_GetREVID() == 0x1001) {
    /* Enable the Flash prefetch */
    __HAL_FLASH_PREFETCH_BUFFER_ENABLE();
  }

  /* Ensure CCM RAM clock is enabled */
  __HAL_RCC_CCMDATARAMEN_CLK_ENABLE();

}
}

}

The sketch did not compiled.
i read forum too but nothing help. 
https://www.stm32duino.com/viewtopic.php?t=189

i want cpu can change clock while running not Once time.

fpistm, 2020-10-30:
I'm not surprise it does not compile as you define a function inside another one....