stm32duino/Arduino_Core_STM32 #1222: USART3 on STM32 printing garbage

arthurkafer, 2020-10-29:
I'm using a STM32F070CBT6TR in a board and I'm trying to use USART3 for debugging. I made a simple configuration to read and send data through it but it only prints garbage on the serial terminal. 

Serial configuration:
Baud rate: 9600
Data bits: 8 bits
Stop bits: 1
Parity: none
Flow control: none

I'm using it as a NFC reader, and it's working perfectly, but I want to debug other things by serial. Here's my code example to use USART3:

```
void setup() {
  // put your setup code here, to run once:
    Serial.setTx(PB10);
    Serial.setRx(PB11);
    Serial.begin(9600);
    delay(2000);
    Serial.println("started");
}

int cnt = 0;
void loop() {
  // put your main code here, to run repeatedly:
    Serial.println(cnt);
}
```

And I have this on my serial monitor: 

`ŽƒŸ[03]B‚ž[13]‚œrl€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L@[02]€L`[03]€L`[02]€L@[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[03]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L@[02]€L`[03]€L`[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L@[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€l`[03]€L`[02]€L@[02]€L`[03]€L`[02]€L`[03]€L`[02]€L`[03]€L`ÿ`

Board configurations: (using this board library)
Board: Nucleo-64
Board Part Number: Nucleo F072RB
U(S)ART Support: "Enabled (generic 'Serial')"
USB support: "None"
USB Speed: "Low/Full Speed"
Optimize: "Smallest (-Os default)"
C Runtime Library: Newlib Nano (default)
Upload Method: STM32CubeProgrammer (SWD)

GPIOS and NFC work fine with this configuration.

Is there something that I forgot to set or a wrong config? 

Thanks in advance

fpistm, 2020-10-29:
Hi,
Just one question you used the Nucleo F072RB target for your custom board based on a STM32F070CBT6TR ?

arthurkafer, 2020-10-29:
Yes, I couldn't find one that was exactly how I needed

ABOSTM, 2020-10-30:
Hi @arthurkafer I tested on NUCLEO_F072RB (officially supported in this Arduino core) and it is working fine with your sketch on PB10 (TX).
The garbage you have, is typical when setting the wrong baudrate in  serial monitor.
Or maybe something with your custom board. But I can help with that.

fpistm, 2020-10-30:
You probably have an hardware issue or config. You can try also to redefine the clock config. Anyway we as mentioned above we could not help on that.

closed 2020-10-30 by fpistm

arthurkafer, 2020-10-30:
It's ok, I have already tested different serial configs and I manage to make serial work by changing compile settings in Arduino, it recognized HardwareSerial for PB10 and PB11 pins. Now I'm changing the project to use CubeIDE because of other libraries and it's working fine by now. 

Anyway, thanks for the help from you!