stm32duino/Arduino_Core_STM32 #1223: Issue with HSE_STARTUP_TIMEOUT and hal_conf_extra.h

DanielLiebler, 2020-10-30:
**Describe the bug**

When HAL_RCC_OscConfig() is called for the initialization, the values of hal_conf_extra.h are not yet present.

This issue comes up because I am using an STM Nucleo F746ZG, powered by an external source. Because the HSE Oscillator quartz is not included in the Nucleo, it is using the ST-Link as a Clock source. But this takes longer to give a clock signal. Therefore I need to increase the HSE_STARTUP_TIMEOUT(This in itself could be changed for the Nucleo default settings, in my opinion). 

**To Reproduce**

I wrote a hal_conf_extra.h containing:

```
#define HSE_STARTUP_TIMEOUT 20000 
#define TestVal
```

I modified HAL_RCC_OscConfig() (In file \system\Drivers\STM32F7xx_HAL_Driver\Src\stm32f7xx_hal_rcc.c) to also contain:

```
#ifndef TestVal
  while(1) {}
#endif
```

Then I connected a debugger and let it reboot.
After a bit of time, I halt the processor with the debugger to see what it is executing, and we are in this infinite loop.

Another possibility to test it would be to check HSE_TIMEOUT_VALUE which should be derived from HSE_STARTUP_TIMEOUT.
You could test this in stm32f7xx_hal_rcc.c like:

```
if (HSE_TIMEOUT_VALUE != 20000) {
  while(1) {}
}
```

And in this case, we are also stuck in the infinite loop.

But the weird thing is:

when running this and having the following code in my Arduino setup() function(having the infinite loops removed):

```
#ifdef TestVal
  Serial.println("TestVal is defined");
#endif
```

The Serial output appears on the console.

**Expected behavior**
The expected behavior would be for those values to be defined like assumed, so we don't enter the infinite loops.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - PlatformIO, ST STM32@8.1.0
 - Upload method: STLink

**Board (please complete the following information):**
 - Name: STM Nucleo F746ZG

**Additional info**

I think this issue is rather related to this repo than the PlatformIO repo, so I posted it here.

Also, if you wonder about the other clock issue, why I even want to increase the startup time: This Issue only arises, when powering up from external power. When I have the USB cable attached or reboot with the reboot button, the clock is fast enough established/already established.

DanielLiebler, 2020-10-30:
Also,
PlatformIO ST STM32@8.1.0
uses framework-arduinoststm32@4.10900.200819 (1.9.0)
which should be this repo

fpistm, 2020-10-30:
Hi @DanielLiebler 
Using `hal_conf_extra.h` with Arduino IDE to redefine `HSE_STARTUP_TIMEOUT ` works as expected.
I already answer to a similar redefinition issue using PIO. I don't know how PIO handle the build and why this not work, it have to be fixed in PIO build process. I even don't know if PIO manage properly the build_opt.h file.
Here, PIO is not supported only Arduino IDE so I close this issue as invalid.

closed 2020-10-30 by fpistm