stm32duino/Arduino_Core_STM32 #1226: PX_N!=PXN

MapleHan, 2020-11-01:
https://github.com/stm32duino/Arduino_Core_STM32/blob/1faf0a0ead7b6ae07258d706c1e9052de0be6cf0/variants/NUCLEO_F429ZI/variant.h，Here PX_N!=PXN,example PB_10!=PB10,But in a generic development board  PB_10==PB10

fpistm, 2020-11-01:
Hi @MapleHan , I don't understand your issue.
Could you explain?

MapleHan, 2020-11-01:
NUCLEO_F419ZI: PB10==36  PB_10==26
      digitalWrite(PB10,HIGH);//PortB10 HIGH
      digitalWrite(PB_10,HIGH);//PortB6 HIGH
Generic_103: PB10==26  PB_10==26
      digitalWrite(PB10,HIGH);//PortB10 HIGH
      digitalWrite(PB_10,HIGH);//PortB10 HIGH

fpistm, 2020-11-01:
Right, there is no issue.
PYn is a pin number and PY_n is a PinName which is a combination of GPIO port and GPIO pin.
For Arduino API pin number is used.  The fact it is equal onthe generic f103 is not an issue, it is a coincidence.

closed 2020-11-01 by fpistm