stm32duino/Arduino_Core_STM32 #1231: COM port not recognised after reset of board.

Kamoool, 2020-11-03:
Hi,

I'm new to STM32 and I need some of it features to work, most of them seems to work very well, but i can't get VCP to work reliably. I'm using nucleo32 L432KC board.

I've got problem with not recognised VCP (straight from L432 chip, not through STLink) after chip reboot.
Things I'm doing:
1. Connecting board NRST to 3.3V pin.
2. Connecting USB_D+ to D2, USB_D- to D10 on board.
3. Connecting USB_5V to 5V onboard and USB_GND to GND.
4. Connecting BOOT0 to 3.3V.
5. Plugging in USB connector to PC.
6. Windows recognises that as STM32 BOOTLOADER DFU.
7. Uploading sketch that uses Serial to board with STM32CubeProgrammer(DFU) upload method.
8. When uploading is finished chip reboots and Windows recognises that as "STMicroelectronics Virtual COM Port (COM#)", and Serial monitor works as intended.
9. I reboot chip either by unpluging and pluging NRST cable or reconnecting USB cable to PC.
10. Now Windows doesn't recognise VCP and it throws Code 43 error, with VID 0000 and PID 0002, so it seems like L432 is not reporting VID and PID correctly after reboot.

To make serial working again i have to again reupload my code and it works until next reboot.

Maybe thats going into different operating mode after reboot?

Any help appreciated.

fpistm, 2020-11-04:
Hi @Kamoool 

As the Nucleo L432KC has no USB connector, the USB clock is not configured.
So, when you flash over DFU, the bootloader properly initialized the USB clock then jump to the user application. That's why it works.
When you reboot, the system clock is to the default state and so the USB clock is not initialized. 

You can override the default `SystemClock_Config(void)` by adding at the beginning of your sketch this one which enable the USB clock, I've tested and it works as expected.

```C++
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};
  RCC_PeriphCLKInitTypeDef PeriphClkInit = {};

  /* Initializes the RCC Oscillators according to the specified parameters 
   * in the RCC_OscInitTypeDef structure.
   */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_MSI;
  RCC_OscInitStruct.MSIState = RCC_MSI_ON;
  RCC_OscInitStruct.MSICalibrationValue = 0;
  RCC_OscInitStruct.MSIClockRange = RCC_MSIRANGE_6;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_MSI;
  RCC_OscInitStruct.PLL.PLLM = 1;
  RCC_OscInitStruct.PLL.PLLN = 40;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV7;
  RCC_OscInitStruct.PLL.PLLQ = RCC_PLLQ_DIV2;
  RCC_OscInitStruct.PLL.PLLR = RCC_PLLR_DIV2;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    Error_Handler();
  }
  /* Initializes the CPU, AHB and APB buses clocks */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_4) != HAL_OK) {
    Error_Handler();
  }
  PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_USB;
  PeriphClkInit.UsbClockSelection = RCC_USBCLKSOURCE_PLLSAI1;
  PeriphClkInit.PLLSAI1.PLLSAI1Source = RCC_PLLSOURCE_MSI;
  PeriphClkInit.PLLSAI1.PLLSAI1M = 1;
  PeriphClkInit.PLLSAI1.PLLSAI1N = 24;
  PeriphClkInit.PLLSAI1.PLLSAI1P = RCC_PLLP_DIV7;
  PeriphClkInit.PLLSAI1.PLLSAI1Q = RCC_PLLQ_DIV2;
  PeriphClkInit.PLLSAI1.PLLSAI1R = RCC_PLLR_DIV2;
  PeriphClkInit.PLLSAI1.PLLSAI1ClockOut = RCC_PLLSAI1_48M2CLK;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK) {
    Error_Handler();
  }
  /* Configure the main internal regulator output voltage */
  if (HAL_PWREx_ControlVoltageScaling(PWR_REGULATOR_VOLTAGE_SCALE1) != HAL_OK) {
    Error_Handler();
  }
}
```

Kamoool, 2020-11-04:
Hello @fpistm,

I've just tested that and it works everytime :) Thank You very much :)

fpistm, 2020-11-04:
Yes, The core is MCU oriented. Variant is mainly a pins mapping description.
You can even create your own variant based on the core that's why several generic target exists. I'm currently working to unified this and make it easy to create new variant.
I close this issue as it is answered. For dedicated support please use the forum.

closed 2020-11-04 by fpistm

MasteringTheMess, 2024-01-19:
Just ran into this issue as well

Thank You, saved me a lot of time - I suppose :)