stm32duino/Arduino_Core_STM32 #1232: can't initialize RPlidar through Serial2 STM32F407VGT6 blackboard

adezie, 2020-11-05:
When reporting any issue, please try to provide all relevant information:

so recently I've bought rplidar a1m8 for my project, but I got some problem, I've initialized it to use serial2, but it wont start anything whatsoever, the pwm ctl i connected it to pc8

**To Reproduce**
user simple connect on rplidar library using serial2

**Expected behavior**
just stucked on serial.begin, without starting main code in void loop()

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
stucked at serial begin so it initialized USB as serial, but it wont start the rplidar protocol

**Desktop (please complete the following information):**
 - OS: WINDOWS
 - Arduino IDE version: newest
 - STM32 core version: newest
 - Tools menu settings if not the default: USB cdc(generic supersede usart), generic serial support
 - Upload method: STlink

**Board (please complete the following information):**
 - Name: BLACK STM32F407VGT6
 - Hardware Revision: VER 3.0
 - Extra hardware used if any: RPLIDAR A1M8

**Additional context**
Add any other context about the problem here.

fpistm, 2020-11-05:
Hi @adezie 

By default the Serial is mapped to Serial1 (PA10/PA9).
I saw in your screenshot that you give to the RPLIDAR  lidar the Serial2 instance. But it not instantiate, it is only declared as extern.
You have to instantiate it properly: https://github.com/stm32duino/wiki/wiki/API#hardwareserial

`HardwareSerial Serial2(PA3, PA2);`

I guess you used this library: https://github.com/robopeak/rplidar_arduino

Anyway I don't understand how it can be build as it is only declared as extern.
```log
simple_connect.ino.cpp.o: in function `setup':
simple_connect.ino.cpp:(.text.setup+0x18): undefined reference to `Serial2'
collect2.exe: error: ld returned 1 exit status

```

adezie, 2020-11-05:
thanks for the reply @fpistm i'll try it tonight, but still U(S)ART and serial configuration in stm32duino quiet confuse me

adezie, 2020-11-05:
@fpistm just try it by now, there's still no sign of success, the lidar won't start and no sign of successful communication between my board with the lidar

fpistm, 2020-11-05:
Please, try the Serial2 without the library.

adezie, 2020-11-05:
[image]
here's the PA2 tx pin after i send "test" data, it works like a charm, hmm does the library come up with something that stm32duino doesn't support yet ?

fpistm, 2020-11-05:
HardwareSerial from the STM32 core is fully compatible with the Arduino API.

adezie, 2020-11-05:
[image]
@fpistm here's the PA2 pin when I run with the library

fpistm, 2020-11-05:
Well, I don't know what I can tell you.
It seems issue come from the library. 
In the library issue there are 2 issues asking if the library is compatible with Due and Yun. So it seems it does not work for all board by default.
You should open an issue to the library.
As the Serial works as expected, I close this issue as it relies to third party library. If you found a core issue then we could reopen this one.

closed 2020-11-05 by fpistm

adezie, 2020-11-05:
actually, i've tested the code using F103 series, it worked, so I think there's still problem on U(S)ART driver on F407

fpistm, 2020-11-05:
This is the same. So it is an Hardware issue.

fpistm, 2020-11-06:
What is the sketch you used ?

adezie, 2020-11-06:
@fpistm i used stm32duino core that used leaflabs documentation and it worked, I've tried using stm32 core from this github and same, no success

fpistm, 2020-11-06:
@adezie 
What is the sketch you used?

adezie, 2020-11-06:
@fpistm i used default simple_connect.ino RPLidar Library

fpistm, 2020-11-06:
Do you know the Serial bandwidth ? 
By default Rx and Tx are 64 bytes.
Could you try to increase both buffer to 256, please?
https://github.com/stm32duino/wiki/wiki/Custom-definitions#serial-rxtx-buffer-size

adezie, 2020-11-06:
@fpistm ahh i see, it could be the bandwidth, here's the link https://github.com/mcauser/MCUDEV_DEVEBOX_F407VGT6 
, I'll try it this morning and send report soon

fpistm, 2020-11-06:
Libmaple core buffer size is 256:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/59227d1ecab7e931a390119c20655f1c4147a668/STM32F4/cores/maple/libmaple/usart.h#L234-L240