stm32duino/Arduino_Core_STM32 #1235: Problem with Programming my STM32F411RE custom board

Vicky-S, 2020-11-08:
**Desktop (please complete the following information):**
 - OS:  Windows
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9.0

**Board (please complete the following information):**
 - Name: STM32F411RE

I have a custom board with STM32F411RE and i can program the board via STLINK. Im exporting the Binary file in arduino and then Uploading it via stlink utility software. 

Im trying to read the status of a GPIO pin then switching on the relay according to it. But while uploading the code its not working. But with the same code if i generate binary file with 1.6.1 core I can able to control the board as desired. What may be the problem? Please Help me to solve the issue.

fpistm, 2020-11-08:
Hi @Vicky-S 
Which target you used ?
I guess you used the Nucleo F411RE target ? as the Generic F411RE is available only since 1.9.0.

By default, now the Nucleo F411RE used the HSE bypass  then I guess your custom board doesn't have this way

Try to use the GenericF411RE target which try several clock config.

Vicky-S, 2020-11-08:
Ok. Let me check it with Generic F411RE.

Vicky-S, 2020-11-08:
In Generic F411RE its working. Is there any difference in UART or SPI By using this Generic F411RE As board?

fpistm, 2020-11-08:
Yes, as the pin mapping is not the same the default instance are probably not the same.
You can check in the variant:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/Generic_F411Rx/variant.h

closed 2020-11-08 by fpistm

Vicky-S, 2020-11-14:
I tried to use SPI2 of STM32F411RE  by selecting Generic STM32F411RE as target but the processor is getting hanged if i try to transfer some data in SPI.

Code to Reproduce.

`#include <SPI.h>
#include <LiquidCrystal.h>
//#include "filter.c"

const int rs = PA15, en = PC8, d4 = PC9, d5 = PC10, d6 = PC11, d7 = PC12;
LiquidCrystal lcd(rs, en, d4, d5, d6, d7);

SPIClass SPI_TWO(PB15, PB14, PB13); //SPIClass SPITWO(MOSI, MISO, SCK);

void setup()
{
  Serial.begin(38400);
  pinMode(PB12, OUTPUT);                   //chip enable output line

  lcd.begin(16, 2);
  lcd.print("InputTest 123...");
  delay(1000);

  SPI_TWO.begin();
  Serial.println("SDC Initialised successfully....");
  lcd.setCursor(1, 1);
  lcd.print("Welcome     ");
  delay(2000);
  lcd.clear();
}

int counter = 1;
void loop()
{
  digitalWrite(PB12, LOW);
  SPI_TWO.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE1) ); // 2 MHz
  SPI_TWO.transfer(0x00);
  SPI_TWO.endTransaction();
  digitalWrite(PB12, HIGH);
  lcd.setCursor(0, 1);
  lcd.print(String(counter++));
  delay(100);
}
`
counter value is not updating in loop. If i remove those SPI functions in loop then its working.

fpistm, 2020-11-14:
https://github.com/stm32duino/Arduino_Core_STM32/blob/caa8a28b281d14f91fbdda3744160fb5d40ca419/variants/Generic_F411Rx/PeripheralPins.c#L203

uncomment this line.

Soon all pins will be available

Vicky-S, 2020-11-14:
Ok. Thank You. Its working Now. 

Do i need to change anything for UARTs and I2C?