stm32duino/Arduino_Core_STM32 #1240: COM port not recognised after reset of board.

FelipeIslas, 2020-11-17:
Hi,

I saw that this Issue was already solved (It's the problem that the COM port after resetting the board isn't recognized) with the nucleo32 L432K card, I have the same problem but for a board that I designed with an STM32L4P5. In this case, is it the same solution?

fpistm, 2020-11-17:
Hi @FelipeIslas 
So if this is a USB com port probably yes. But there is no STM32L4P5 based board variant defined in the core so you have to properly create/update your variant with a proper clock config.

FelipeIslas, 2020-11-17:
Ok, Thanks for comment. I'm relatively new defining variants, I'm just learning. So I can get a proper clock config from STMCubeMx rigth?

fpistm, 2020-11-17:
> 
> 
> Ok, Thanks for comment @fpistm. I'm relatively new defining variants, I'm just learning. So I can get a proper clock config from STMCubeMx rigth?

Yes

fpistm, 2020-11-17:
Please use the forum for question and support: https://www.stm32duino.com

closed 2020-11-17 by fpistm