stm32duino/Arduino_Core_STM32 #1241: problem with libraries

Mstitelp, 2020-11-18:
the code worked fine. there was a failure two days ago. now throws an error with all 
libraries.`C:\Users\Admin\Documents\Arduino\libraries\TinyGPSPlus-1.0.2b\src\TinyGPS++.cpp: In member function 'bool TinyGPSPlus::encode(char)':
C:\Users\Admin\Documents\Arduino\libraries\TinyGPSPlus-1.0.2b\src\TinyGPS++.cpp:63:12: warning: this statement may fall through [-Wimplicit-fallthrough=]
   63 |     parity ^= (uint8_t)c;
      |     ~~~~~~~^~~~~~~~~~~~~
C:\Users\Admin\Documents\Arduino\libraries\TinyGPSPlus-1.0.2b\src\TinyGPS++.cpp:64:3: note: here
   64 |   case '\r':
      |   ^~~~
In file included from C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_tim_ex.c:5:
C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c: In function 'HAL_TIMEx_RemapConfig':
C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:60: warning: unused parameter 'htim' [-Wunused-parameter]
 1734 | HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:75: warning: unused parameter 'Remap' [-Wunused-parameter]
 1734 | HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
      |                                                                  ~~~~~~~~~^~~~~
Скетч использует 39568 байт (60%) памяти устройства. Всего доступно 65536 байт.
Глобальные переменные используют 2340 байт (11%) динамической памяти, оставляя 18140 байт для локальных переменных. Максимум: 20480 байт.
`

Mstitelp, 2020-11-18:
same problem with examples

fpistm, 2020-11-18:
Hi @Mstitelp 
Could you be more precise?
Which library? Which target? What is the problem? The code compile? Only warnings? 
All relevant information will be appreciated.
Thanks

Mstitelp, 2020-11-18:
`In file included from C:\Users\Admin\Desktop\autopilot\______________\AutopilotFinishVersion\AutopilotFinishVersion.ino:1:
C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\EEPROM\src/EEPROM.h:247:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
  247 | static EEPROMClass EEPROM;
      |                    ^~~~~~
C:\Users\Admin\Documents\Arduino\libraries\IBusBM-master\src\IBusBM.cpp: In member function 'void IBusBM::begin(HardwareSerial&, TIM_TypeDef*, int8_t, int8_t)':
C:\Users\Admin\Documents\Arduino\libraries\IBusBM-master\src\IBusBM.cpp:85:74: warning: unused parameter 'rxPin' [-Wunused-parameter]
   85 | void IBusBM::begin(HardwareSerial &serial, TIM_TypeDef * timerid, int8_t rxPin, int8_t txPin) {
      |                                                                   ~~~~~~~^~~~~
C:\Users\Admin\Documents\Arduino\libraries\IBusBM-master\src\IBusBM.cpp:85:88: warning: unused parameter 'txPin' [-Wunused-parameter]
   85 | void IBusBM::begin(HardwareSerial &serial, TIM_TypeDef * timerid, int8_t rxPin, int8_t txPin) {
      |                                                                                 ~~~~~~~^~~~~
C:\Users\Admin\Documents\Arduino\libraries\TinyGPSPlus-1.0.2b\src\TinyGPS++.cpp: In member function 'bool TinyGPSPlus::encode(char)':
C:\Users\Admin\Documents\Arduino\libraries\TinyGPSPlus-1.0.2b\src\TinyGPS++.cpp:63:12: warning: this statement may fall through [-Wimplicit-fallthrough=]
   63 |     parity ^= (uint8_t)c;
      |     ~~~~~~~^~~~~~~~~~~~~
C:\Users\Admin\Documents\Arduino\libraries\TinyGPSPlus-1.0.2b\src\TinyGPS++.cpp:64:3: note: here
   64 |   case '\r':
      |   ^~~~
In file included from C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_tim_ex.c:5:
C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c: In function 'HAL_TIMEx_RemapConfig':
C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:60: warning: unused parameter 'htim' [-Wunused-parameter]
 1734 | HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~
C:\Users\Admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:75: warning: unused parameter 'Remap' [-Wunused-parameter]
 1734 | HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
      |                                                                  ~~~~~~~~~^~~~~
Скетч использует 39088 байт (59%) памяти устройства. Всего доступно 65536 байт.
Глобальные переменные используют 2324 байт (11%) динамической памяти, оставляя 18156 байт для локальных переменных. Максимум: 20480 байт.
`

initially the sketch compiled without problems and I worked with it for two months. then the cursor ran across the monitor, the lines of code were deleted and the computer hung. I rebooted and started having problems with the libraries. warnings are issued and after filling in the code, the microcontroller does not work. I tried to completely reinstall everything, the result is zero. I do not know what to do. can the virus climb ?? 
libraries IBus, TinyGPS, Ultrasonic, Servo, Navigation. I will try on another computer. sorry if not exactly, I tried my best))) thanks for your feedback

Mstitelp, 2020-11-18:
it is not clear, once it is compiled with warnings, then normally without warnings. I'm already blowing my head))

fpistm, 2020-11-19:
Hi,

Warning are shown only the first time you build then the cache is used and so the warning are no more displayed.

For the one in the core:
This is simply sole unused parameters. They are linked to the HAL drivers and so will not fix them in the core. They will be fixed in the HAL driver repository.
Other are linked to third party library and should be fixed there.

Finally about: 
`exec: "/bin/arm-none-eabi-g++": file does not exist`

Without more information I can not help.

Mstitelp, 2020-11-19:
figured it out !!! this is a problem with my computer. works fine on another computer. I will look for. thanks for the tip !!!!!!!!!

fpistm, 2020-11-19:
Probably corrupted package.

closed 2020-11-19 by fpistm