stm32duino/Arduino_Core_STM32 #1245: [CI] PlatformIO test

fpistm, 2020-11-30:
Action PlatformIO test failed since few days.

Ex: https://github.com/stm32duino/Arduino_Core_STM32/pull/1242/checks?check_run_id=1469456447

```
Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1257, in invoke
    sub_ctx = cmd.make_context(cmd_name, args, parent=ctx)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 700, in make_context
    self.parse_args(ctx, args)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1048, in parse_args
    value, args = param.handle_parse_result(ctx, opts, args)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1630, in handle_parse_result
    value = invoke_param_callback(self.callback, ctx, self, value)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 123, in invoke_param_callback
    return callback(ctx, param, value)
  File "/usr/local/lib/python3.8/dist-packages/platformio/commands/project.py", line 115, in validate_boards
    pm.board_config(id_)
  File "/usr/local/lib/python3.8/dist-packages/platformio/package/manager/platform.py", line 185, in board_config
    for manifest in self.get_installed_boards():
  File "/usr/local/lib/python3.8/dist-packages/platformio/util.py", line 51, in wrapper
    self.cache[key] = (time.time(), func(*args, **kwargs))
  File "/usr/local/lib/python3.8/dist-packages/platformio/package/manager/platform.py", line 161, in get_installed_boards
    for config in p.get_boards().values():
  File "/github/home/.platformio/platforms/ststm32/platform.py", line 115, in get_boards
    result[key] = self._add_default_debug_tools(result[key])
  File "/github/home/.platformio/platforms/ststm32/platform.py", line 159, in _add_default_debug_tools
    assert debug.get("openocd_target"), (
AssertionError: Missed target configuration for mkr_sharky

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

@valeros could you help on this. It seems there is a regression and don't know why. I guess something changed on PIO side.
Thanks in advance.

valeros, 2020-11-30:
Thanks for pinging me, should be fixed now, please restart your CI.

fpistm, 2020-11-30:
Thanks @valeros for your fast support. it works now.

closed 2020-11-30 by fpistm