stm32duino/Arduino_Core_STM32 #1254: Native i2c doesn't work on STM32F103 Blue Pill

drelephant, 2020-12-04:
I'm running v 1.9.0 STM32 Core on Arduino 1.8.5 on a Blue Pill STM32F103.

I'm trying to run the i2c scanner Example, but it never finds any devices. I've looked with an oscilloscope at both lines and there is no activity.

SDA is connected to PB7, SCL to PB6. Both lines have a 4k7 pullup.

The devices are found when I use SoftWire with this sketch:

```
#include <SoftWire.h>

//SoftWire SWire(PB6, PB7, SOFT_FAST);
SoftWire SWire(PB7, PB6);  // (sdaPin, sclPin)

void setup() {
  delay(5000);
  Serial.begin(9600);
  SWire.begin();
  Serial.println("\nSoftware I2C.. Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    SWire.beginTransmission(address);
    error = SWire.endTransmission();
   
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.println(address, HEX);
    }   
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");

  delay(500);           // wait 5 seconds for next scan
}
```
Shows:
```
Scanning...
I2C device found at address 0x27
I2C device found at address 0x68
done
```

But with the i2c Scanner example:
```
// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//

/* Example pinmap for Bluepill I2Cs (by Testato)

 I2C-1 standard pins: PB7(sda) PB6(scl)
 Use it by "Wire" without pin declaration
  Wire.begin();

 I2C-1 alternative pins: PB9(sda) PB8(scl)
 Remap the first I2C before call begin()
  Wire.setSDA(PB9);
  Wire.setSCL(PB8);
  Wire.begin();

 I2C-2: PB11(sda) PB10(scl)
 Remap the second I2C before call begin()
  Wire.setSDA(PB11);
  Wire.setSCL(PB10);
  Wire.begin();

 If you want to use the two I2Cs simultaneously, create a new instance for the second I2C
  TwoWire Wire2(PB11,PB10);
  Wire2.begin();
 
*/

#include <Wire.h>

void setup() {

  Serial.begin(9600);
  
  Wire.begin();
  Wire.setClock(100000);
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");

  delay(500);           // wait 5 seconds for next scan
}
```
, no devices are found:
```
No I2C devices found
Scanning...
No I2C devices found
Scanning...
```

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Arduino IDE version: 1.8.5
 - STM32 core version: 1.9.0
 - Upload method: Maple DFU Bootloader 2.0

**Board:**
 - Name: Blue Pill STM32F103

fpistm, 2020-12-04:
Hi @drelephant 
Are you sure you used the STM32 core ? SoftWire does not exist for this core but for Roger's one (Libmaple core).
I2C on BP works fine with this core. For Roger's core there is a issue about this: 
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/799
and related PR:https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/804

drelephant, 2020-12-04:
Thanks for the reply @fpistm - no, I'm not sure, I'm not an expert. How can I check?

drelephant, 2020-12-04:
Does this help at all?

[image]

fpistm, 2020-12-04:
Yes this help. So you well used the STM32 core.
Now, is your BP a real STM32F103C8 there is a lot of fake?
I have no other clue as the I2C scanner works well on my side and it seems your setup is correct as you used PU.
About the SoftWire, I was not aware of this library: https://github.com/stevemarple/SoftWire

drelephant, 2020-12-04:
After an entire night of trying, I eventually fixed it by using the ST-Link to put the latest bootloader on the blue pill.

[https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13.bin](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13.bin)

Not sure why it wouldn't work with the old one, but I'm happy that ordeal is over.

After that it worked immediately, and now I can get some sleep... Thanks for your help @fpistm .

closed 2020-12-04 by drelephant

vladyslav-savchenko, 2021-01-16:
Hi guys!
After patch **1ac0186**  i2c scanning hangs on after non existing address (Board Generic stm32F103C series).
But worked  before.
Looks like after NACK error
`if (sel_hard->error_flags & I2C_SR1_AF) { /* NACK */`
Wire does not reset the bus and freezes.

fpistm, 2021-01-16:
Hi @vladyslav-savchenko 
what is the patch you refers: 1ac0186  ?

Alchemyisthis, 2024-04-08:
got need to gno data , i2c scanner works on a older module computer, even when activated on the other newer module computer but does not work when the code is uploaded from the newer computer, ps the i2c works for on the chip its just the scan that freezes....

Alchemyisthis, 2024-04-10:
> got need to gno data , i2c scanner works on a older module computer, even when activated on the other newer module computer but does not work when the code is uploaded from the newer computer, ps the i2c works for on the chip its just the scan that freezes....

ok guys my issue got fixed, but i dont want to try and recreate the issue so i dont gno what i did that really worked, the issue is not in the code but in the environment, i believe it could be a variety of factors that could trigger the same effect, but i presume that the issue due to the wire.h or what ever wire your using might be out dated or the environment is compared to the wire requirements, so long story sort delete Arduino , reset the computer , reinstall and configure your environment verbose on compile an verbose on upload , check anything that relates to wire, the verbose should give you a int as to what is the real issue if you are adept ...