stm32duino/Arduino_Core_STM32 #1263: A fatal error has been detected by the Java Runtime Environment:

Proxcentaur, 2020-12-14:
I have been using STM32F1C series since 5 days ago. For the first time using this board, I have successfully upload a blink program both of using FTDI and STM32 bootloader through USB port. But today I got a strange error like this:
**
maple_loader v0.1
Resetting to bootloader via DTR pulse
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=5852, tid=1516
#
# JRE version: Java(TM) SE Runtime Environment (15.0.1+9) (build 15.0.1+9-18)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (15.0.1+9-18, mixed mode, sharing, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\dartmedia\Documents\Arduino\hardware\Arduino_STM32\tools\win\hs_err_pid5852.log
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.**

Why did this happen? I have updated STM32 boards in Arduino IDE and my JDK to the last version but the error still occurs.

fpistm, 2020-12-14:
Hi @Proxcentaur .
You don't use this core but this one:

https://github.com/rogerclarkmelbourne/Arduino_STM32

Open an issue in this repo. Thanks.

closed 2020-12-14 by fpistm

babipsylon, 2021-02-27:
unsubcribing does not seem to work from this thread