stm32duino/Arduino_Core_STM32 #1266: [New Variant] STM32H743VIT6 (and STM32H750VBT6)

rtek1000, 2020-12-19:
All info here: https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M

[image]

rtek1000, 2020-12-19:
USD $23.00

https://pt.aliexpress.com/item/1005001543610862.html?spm=a2g0s.9042311.0.0.7edfb90aBErHQq

rtek1000, 2020-12-19:
Note: Attention!!! Someone looking to use the STM32H7 for pins and high frequency output: The STM32H7 did only 20MHz (PA0) under the 480MHz clock. While the STM32F407VGT6 did 33MHz under the 168MHz clock.

fpistm, 2021-02-12:
Added to the [New Variant] request list #722.

closed 2021-02-12 by fpistm

ag88, 2021-11-21:
note that the variant made for WeAct MiniSTM32H743VITX would likely work for MCUDev DevEBox STM32H743VIT6
https://github.com/stm32duino/Arduino_Core_STM32/pull/1552
As it uses 25 Mhz HSE crystal same on both boards.
However, do note that the symbols LED_BUILTIN and USER_BTN are different between the 2 boards.
To use the WeAct variannt for MCUDev DevEBox STM32H743VIT6, avoid using the 2 symbols or re-define them in your arduino sketch. you could also simply clone the WeActH743VITX variant file and make mods locally e.g. make a new variant if you prefer.

PR #1573 (2021-11-30): variants: add DevEBox H743VITX and H750VBT support
https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M

Fixes #1145 and fixes #1266
commit f5c6e7ac (2021-11-30): variants: add DevEBox H743VITX and H750VBT support

https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M

Fixes #1145 and fixes #1266