stm32duino/Arduino_Core_STM32 #1272: HardwareTimer: No method to set master mode selection

darkstar007, 2020-12-30:
In order to trigger ADC's at a regular interval, TRGO output from a Timer is often used. The currect class definition of the HardwareTimer does not allow this to happen. The easiest way for this is to add a function/method that does something along the lines of
```
enableTRGO() {
   TIM_MasterConfigTypeDef mct;
    mct.MasterOutputTrigger = TIM_TRGO_UPDATE;
    mct.MasterSlaveMode = TIM_MASTERSLAVEMODE_ENABLE;
    HAL_TIMEx_MasterConfigSynchronization(&s_TimerInstance, &mct);
}
```

ABOSTM, 2021-01-04:
Hi @darkstar007,
Happy new year !!
This is right, HardwareTimer API doesn't provide all capabilities of STM32 timer hardware. It provides only basic usage of timer.
Nevertheless HAL API can be used directly in Arduino sketch, so that experts have access to all capabilities of STM32 timer hardware.

Tackling timer master mode is far more complex that you suggested:
* all timer instances do not support master configuration
* master management would come with slave management (which imply management of triggers )
* you configured TRGO, but some STM32 family have some instances with TRGO and TRGO2
* you configured master in `enable `mode, but there are other modes like `update`, `compare pulse`, ...
* ...

So it would be a huge task, and there is no plan today for such implementation.

I am glad to ear that you manage it on your own. You can even share your code to others : 
https://www.stm32duino.com/viewforum.php?f=42

fpistm, 2021-05-03:
As stated by @ABOSTM 

> I am glad to ear that you manage it on your own. You can even share your code to others :
> https://www.stm32duino.com/viewforum.php?f=42

It is a user code sharing so I close it as no plan to implement this and:

> This is right, HardwareTimer API doesn't provide all capabilities of STM32 timer hardware. It provides only basic usage of timer.
> Nevertheless HAL API can be used directly in Arduino sketch, so that experts have access to all capabilities of STM32 timer hardware.

Thank you for sharing.

closed 2021-05-03 by fpistm