stm32duino/Arduino_Core_STM32 #1273: STM32G071XX series

msadeghz, 2021-01-09:
Hi friends,
when I see "STM32G071RB" mcu is supported in repo first page, Can I conclude that all STM32G071XX MCUs are supported?

fpistm, 2021-01-09:
Hi,
This means a board based on STM32G071RB is defined.
The core supports the STM32G0 series and several other STM32 series.
If you have another STM32G071 then you will have to define your board (variant) properly (Flash size, RAM, size,...)
In next release it will be easier to add new generic variant.
I close this issue as it is not related to a bug. For support/question please use the forum as explained in the GH issue template.

closed 2021-01-09 by fpistm

msadeghz, 2021-01-09:
> GH issue template.

thank you very much, 
you were right, I have mistaken in tag selection, I had a question,
so, can you give me a link to adding the variant? is below link related to what you said?
[](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))

fpistm, 2021-01-09:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-%28board%29

your best bet is to copy the Nucleo G071RB then update it to your needs.

You can also create it outside the core:
https://github.com/stm32duino/wiki/wiki/Custom-board-based-on-a-core