stm32duino/Arduino_Core_STM32 #1276: Cannot use PG9 as digital pin with NUCLEO-H743ZI2

pierremolinaro, 2021-01-12:
- OS: Mac OS X
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9.0
 - Board Name: NUCLEO-H743ZI2

**Describe the bug**
I try to use PG9 as digital output :
- `pinMode (PG9, OUTPUT) ;` in `setup`;
- `digitalWrite (PE9, HIGH) ;` has no effect, pin is not a high level digital output.

**Bug Fix**
I think the bug is in the `~/Library/Arduino15/packages/STM32/hardware/stm32/1.9.0/variants/NUCLEO_H743ZI/variant.h` file.

Line 174 is : `#define NUM_DIGITAL_PINS  99`, and line 164 is `#define PG9  99`. `PG9` is the last pin, so `NUM_DIGITAL_PINS` should be equal to `100`.

So changing the line 174 to `#define NUM_DIGITAL_PINS  100` solves the bug.

**Note**
I think the line 176 `#define NUM_DIGITAL_PINS  97` (for NUCLEO_H743ZI) should be also changed to `#define NUM_DIGITAL_PINS  98`, for the same reason : last pin is `PG12` (#97).

Best regards,

Pierre Molinaro

fpistm, 2021-01-12:
Hi @pierremolinaro 

you're right it should be 100.
This will be fixed with the #1091

pierremolinaro, 2021-01-12:
Thank you very much. Note also that `NUM_DIGITAL_PINS` should be 98 for NUCLEO_H743ZI.

Best regards,

Pierre Molinaro

closed 2021-04-17 by fpistm