stm32duino/Arduino_Core_STM32 #1277: Cannot use PA6 as analog input with a NUCLEO-H743ZI2

pierremolinaro, 2021-01-12:
- OS: Mac OS X
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9.0
 - Board Name: NUCLEO-H743ZI2

**Describe the bug**
I try to use PA6 as analog input. Calling `analogRead (PA6)` returns always zero.

**Bug Fix**
I think the bug is in the `~/Library/Arduino15/packages/STM32/hardware/stm32/1.9.0/variants/NUCLEO_H743ZI/PeripheralPins.c` file.

Line 57 `{PA_6,  ADC1,  STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 3, 0)}, // ADC1_INP3` is commented.

Uncommenting this line solves the bug.

Best regards,

Pierre Molinaro

fpistm, 2021-01-12:
Hi @pierremolinaro 

In this case this is not a bug. The variant has been described like this.
Anyway with #1091 this will be changed as all pins would be available.

pierremolinaro, 2021-01-12:
Ok, thank you very much for your answer.

Best regards,

Pierre Molinaro

closed 2021-04-17 by fpistm