stm32duino/Arduino_Core_STM32 #1279: Can't use ArduinoModbus with STM32 Blue Pill (Maple Mini clone)

requeijaum, 2021-01-15:
**Please, before reporting any issue**
- [x] Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- [x] Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](http://stm32duino.com)
- [x] Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
- [x] **Avoid** to submit a GitHub issue for project troubleshooting.
 
Any questions/feedback/suggestions should be discussed on the [stm32duino forum](http://stm32duino.com):
  * questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=38)

When reporting any issue, please try to provide all relevant information:

**Describe the bug**
A clear and concise description of what the bug is.

Can't use ArduinoModbus with STM32 Blue Pill (Maple Mini clone).

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Tried ArduinoModbus's sketch examples and a source-code of an already working project using Arduino with ESP32 (DOIT v1 board) and Arduino UNO R3 (ATM328P-PU).

Steps to reproduce the behavior:
1. Plug
2. Press **compile**
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 v2004
 - Arduino IDE version: 1.8.13
 - STM32 core version: 2020.12.26
 - Tools menu settings if not the default: **I don't know what's this.**
 - Upload method: STLink v2 (clone)

**Board (please complete the following information):**
 - Name: Blue Pill
 - Hardware Revision: 64K Flash
 - Extra hardware used if any: Generic RS-485 transceiver module with MAX485.

**Additional context**

See related issue on ArduinoModbus.
https://github.com/arduino-libraries/ArduinoModbus/issues/47

fpistm, 2021-01-15:
Hi @requeijaum 

Regarding the core version you specified:
> STM32 core version: 2020.12.26

 and the log file available in ArduinoModBus library 

> C:\Users\rafae\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.12.26

You do not use the STM32 core but Roger's core:
https://github.com/rogerclarkmelbourne/Arduino_STM32

So you should open an issue there or in the forum: https://www.stm32duino.com

closed 2021-01-15 by fpistm

requeijaum, 2021-01-15:
Thanks for the quick feedback @fpistm !