stm32duino/Arduino_Core_STM32 #1283: Serial USB on STM32G431KB not working

JiriBilek, 2021-01-24:
First I want to thank you for the effort in maintaining this library for us. I like the opportunity to quickly create an app for various STM32 MCUs.
The problem I am going to describe is related to my custom board. 
**If this library and its support is constrained to STM32 Boards only, please feel free to close this issue.**

**My device**
I made a custom board with STM32G431KB. The board works fine with eclipse compiled code not related to Arduino (CubeMX generated). As I realized the board is quite similar to Nucleo-G431KB, I was curious if the board would work with stm32duino. The basic difference between my board and the Nucleo is that I don't solder the oscillator and hence don't use external clock (HSE).

**Describe the bug**
 I am stuck with the USB CDC. The board runs fine in other configurations but selecting any of the two _USB Support (if available): CDC_ options causes the board to freeze on start (it never reaches setup() function) and the PC is reporting unrecognized USB Device.
The board does not freeze when selecting the _HID (keyboard and mouse)_ option but I can't test it if it works somehow :-(

**To Reproduce**
Try any sketch and compile it with _USB Support: CDC_ option.

**Desktop:**
 - OS: Win10
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9.0
 - Tools menu settings if not the default: Nucleo G431KB, U(S)ART Support: Enabled, USB Support: CDC, USB Speed: Low/Full Speed
 - Upload method: STM32CubeProgrammer DFU

**Board:**
 Custom design based on STM32G431KBT6, similar to Nucleo-G431KB except the external oscillator is not soldered on mine board. USB CDC runs fine on code generated with CubeMX.

**Additional context**
The lack of the external oscillator should not be an issue, I learned from the code that this library doesn't use it anyway.
https://github.com/stm32duino/Arduino_Core_STM32/blob/5a0dcfcb8779a3597e57434c26341389b597184c/variants/NUCLEO_G431KB/variant.cpp#L61

I somehow traced the application from the start and found that the clock setup may be wrong. However, after fixing it (code posted below), the execution of the code ends on 
https://github.com/stm32duino/Arduino_Core_STM32/blob/5a0dcfcb8779a3597e57434c26341389b597184c/system/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_pcd.c#L971
From this call the processor never returns. It may be related with the USB enumeration process that's interrupt driven but tracing this is unfortunately over my abilities.

**Fixing the clock setup**
I think the USB Clock domain must be enabled. As I am not able to test it on Nucleo board I am inserting the code here instead of making a PR.

It is replacement of the function:
https://github.com/stm32duino/Arduino_Core_STM32/blob/5a0dcfcb8779a3597e57434c26341389b597184c/variants/NUCLEO_G431KB/variant.cpp#L59

```
/**
  * @brief System Clock Configuration
  * @retval None
  */
void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
  RCC_PeriphCLKInitTypeDef PeriphClkInit = {0};

  /** Configure the main internal regulator output voltage
  */
  HAL_PWREx_ControlVoltageScaling(PWR_REGULATOR_VOLTAGE_SCALE1_BOOST);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI|RCC_OSCILLATORTYPE_HSI48;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.HSI48State = RCC_HSI48_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = RCC_PLLM_DIV4;
  RCC_OscInitStruct.PLL.PLLN = 85;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = RCC_PLLQ_DIV2;
  RCC_OscInitStruct.PLL.PLLR = RCC_PLLR_DIV2;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_4) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the peripherals clocks
  */
  PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_USB;
  PeriphClkInit.UsbClockSelection = RCC_USBCLKSOURCE_HSI48;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK)
  {
    Error_Handler();
  }
}
```

*My other thoughts*
I compared stm32g4xx_hal_pcd.c file with mine generated by MXCube on another project and they are different. I tried to replace this file but it didn't help. It was only a shot in the dark, I don't think it's a good idea to change only one file in the HAL layer.

fpistm, 2021-01-25:
Hi @JiriBilek 
Yes it seems there is an issue with USB.
About the `SystemClock_Config` you can simply ad your own one at sketch level using `extern "C"` as it is a weak function.:
```
extern "C" void SystemClock_Config') {
...
}
```
Unfortunately, I will be able to investigate this soon as I'm busy on other task to release the core 2.0.0.

JiriBilek, 2021-01-25:
@fpistm Thanks for your reply. BTW, I looked at `variant.c` for NUCLEO G431RB and it contains the same `SystemConfig_Clock` function I proposed. But USB Serial didn't work either. The KB and RB MCU should with a bit of luck run the same code.
This is not a time critical task for me, no worries.

fpistm, 2021-01-25:
Right, I've tested with Nucleo G474RE.

JiriBilek, 2021-04-25:
Just update:
Tried on V2.0 hoping changing HAL code would fix it. Unfortunately, the bug persists.
```
Connection Status        : 0x02 (Device failed enumeration)
Port Chain               : 1-2-2-2

Device Manager Problem   : 43 (CM_PROB_FAILED_POST_START)
Used Endpoints           : 0

```

fpistm, 2021-04-28:
Hi @JiriBilek 
I did not expect it works with 2.0.0. As tagged I've planned to fix it for 2.x.x.
I've found the issue. The USB IRQ handler is not called as I forgot to define it for G4 😭.
When I've added the G4 serie I did not have one with USB connector that's why I've didn't see this.

As a workaround you can add this in your sketch to have USB:
```cpp
extern PCD_HandleTypeDef g_hpcd;
extern "C" void USB_LP_IRQHandler(void)
{
  HAL_PCD_IRQHandler(&g_hpcd);
}
```

I will provide soon a fix for this.

JiriBilek, 2021-04-28:
Hi @fpistm ,
thanks a lot for a fix. I confirm, the USB serial now works like a charm.
Closing as nothing more could be done here.

closed 2021-04-28 by JiriBilek

fpistm, 2021-04-28:
Thanks for the feedback.
About closing the issue it would be automatically closed when the fix will be merged.

PR #1374 (2021-04-29): USB reviews and Device library update
* USB: update STM32 USB Device library to [v2.8.0](https://github.com/STMicroelectronics/stm32_mw_usb_device/releases/tag/v2.8.0)
* USB: update core and class files after middleware update
* USB: review WakeUp IRQn and IRQHandler across series
* USB: review IRQn and IRQHandler across series (fix #1283)
* USB: add weaked system clock config from resume
* USB: fix EPO STALL issue in STM32 USB Device library
* UCPD: disable Dead Battery

####  STM32 series tested: 
- [x] STM32F0
- [x] STM32F1
- [x] STM32F2
- [x] STM32F3
- [x] STM32F4
- [x] STM32F7
- [x] STM32 [... cut]
commit a68396f4 (2021-04-29): USB: review IRQn and IRQHandler across series

Fix #1283