stm32duino/Arduino_Core_STM32 #1285: SPI DMA enhanced transfers

ag88, 2021-01-26:
**Is your feature request/improvement related to a problem? Please describe.**
current standard Arduino SPI api doesn't do DMA.
However it is a strong feature on STM32 mcus. 
DMA accelerated transfers are particular useful for e.g. LCDs

**Describe the solution you'd like**
libmaple (roger's) has some additional API calls that are DMA specific
e.g. the dmatransfer() functions and dmasend()

i'm not too sure if it is a good idea to mirror those DMA APIs across into STM core. 
or perhaps they could be implemented in existing enhanced transfer() methods.
hence, i'd leave that decision to you

**Describe alternatives you've considered**
i happened to be making an 'optimised' ILI9341 lcd 'library', not really but more like a 'cleaned up' Adafruit_ILI9341_STM from Roger's core. i think i'd try to make it work in 'both cores'. In Roger's core i'm using the dmaSend() call for the accelerated pixel push. for ST's core i'd use the bulk buffer transfer() functions

**Additional context**
 Arduino SPI api
https://www.arduino.cc/en/Reference/SPI
STM official implementation
https://github.com/stm32duino/wiki/wiki/API#spi
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SPI/src/SPI.h

libmaple (roger's) core implementation
F1
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/SPI/src/SPI.h
F4
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/libraries/SPI/src/SPI.h

ag88, 2021-01-26:
oops, not that simple, even the 'SPI api' there are subtle differences between libmaple and official core and original Arduino implementations. so no hurry really. it'd take time to update my codes as well.

fpistm, 2021-01-26:
Hi @ag88 
yes, the main difference btw Libmaple core and this one is that the STM32 core is fully compatible with the Arduino API and must be kept.

stas2z, 2021-02-12:
Hi, you still can use dma using STM32 HAL api with this core
ive done it some time ago to make a fast library for st7735 which is similar to ili9341 in terms of init and spi proto
you have to modiify SPI library a bit to get access to _spi_handle of SPI object
ive posted part of my dma code at #1035 
you also can use stm32cubemx to generate dma init for your mcu (mine was f401)

ag88, 2021-02-14:
hi thanks, i need to pickup skills on hal and would likely implement a 'trial' fork of spi.h and spi.cpp
my guess is that this would be somewhat different across the series as on a hardware level stm32f4xx has more sophisticated than say stm32f103. i'd likely limit the scope initially to those that i've on hand e.g. the stm32f103 pills, stm32f4 pills

ag88, 2021-03-31:
consider this similar to https://github.com/stm32duino/Arduino_Core_STM32/issues/1035

closed 2021-03-31 by ag88