stm32duino/Arduino_Core_STM32 #1286: trying to add support stm32g070kb uC

msadeghz, 2021-01-26:
I have tried to add stm32g070kb support to the project. I have created the variant steps as is said in wiki, but I see below error during compilation
note that the stm32g070kb doesn't support LPTIM at all but stm32duino mistaken the supportability. when or how can check configuration header files to fix this issue?

```
In file included from C:\Users\Sadegh\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper\src\stm32\timer.c:14:
C:\Users\Sadegh\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper\src\stm32\timer.c: In function 'getTimerUpIrq':
C:\Users\Sadegh\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:79:19: error: 'TIM7_LPTIM2_IRQn' undeclared (first use in this function)
   79 | #define TIM7_IRQn TIM7_LPTIM2_IRQn
      |                   ^~~~~~~~~~~~~~~~
C:\Users\Sadegh\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper\src\stm32\timer.c:404:16: note: in expansion of macro 'TIM7_IRQn'
  404 |         IRQn = TIM7_IRQn;
```

fpistm, 2021-01-26:
Hi @msadeghz 
I guess you try on top of the master?
G0 has some issues and will be corrected with the next HAL/LL update, see #1253 
anyway I'm surprised you got this issue on top of the master with the TIMER.

msadeghz, 2021-01-30:
all right,
can you give me some information to I fix it temporarily.
how do header files works in path `stm32\1.9.0\cores\arduino/stm32/`? how can I edit them based on headers in path `stm32\1.9.0\system\Drivers`

msadeghz, 2021-01-30:
as you see in the pic, stm32g070kb has APB1 and APB2 clock sources:
[image]

but in the hal driver file, `\stm32\1.9.0\system\Drivers\STM32G0xx_HAL_Driver\Inc\stm32g0xx_hal_rcc.h` , APB2 is not defined!
[image]
 
how can I edit the hal driver?

fpistm, 2021-01-30:
In fact this is already solved in the master:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1141

closed 2021-01-30 by fpistm

msadeghz, 2021-01-30:
another question,
in modifying the `PeripheralPins.h` file, how can I calculate `GPIO_AFx_xxxxx` variable?

fpistm, 2021-01-30:
You don't have to modifying yourself the file it is automatically generate.
It is provided thanks the STM32 Tools package.
Or online:
https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap/Arduino/STM32G0

And please use the forum. 😉

msadeghz, 2021-01-30:
thank you very much,
I'm not familiar with the stm32duino ecosystem and forum, I have just googled and didn't find anything.