stm32duino/Arduino_Core_STM32 #1289: Wire's static buffers etc., limits use to one instance even on the many new STM32 devices that contain multiple I2C peripherals

chrisco484, 2021-01-31:
I can understand the rationale for preferring statics back in the 8bit Arduino days but in the STM32 world there is usually a lot more RAM and often multiple UARTs, I2C etc.,

The use of static attributes in the classes that implement support for devices such as I2C seems very restrictive given that it means you can only ever effectively use one instance of the class even though a particular STM32 MCU may have 1, 2 or 3 available "on chip".

Event the humble Blue Pills have a STM32F1* with 2 x I2C peripherals - but how can we use Wire to access the second one? It would override the buffers used by the first instance of wire.

Is there any reason why these static attributes could not be made into object attributes instead of class attributes?

chrisco484, 2021-01-31:
Update:

I must have been looking at old source code - it seems like this was fixed in 1.9.0! Cool!

closed 2021-01-31 by chrisco484

fpistm, 2021-01-31:
Yes it was fixed. We found this more than 1 year  😉